prusa3d/PrusaSlicer #9020: 2.5: extrusion width percentage refers to layer height instead of nozzle size

2Balu2, 2022-10-09:
### Description of the bug

In the past versions the extrusion width settings in % did refer to the nozzle size (Extruder settings from the printer) which was really nice. Now in PrusaSlicer-2.5.0 the width refers to the layer height. I first thought it was Arachne, but it is the same with the Classic Generator.

### Project file & How to reproduce

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+MacOS-arm64-202209060713

### Operating system

MacOS 12.6

### Printer model

Prusa Mini+

neophyl, 2022-10-09:
Its always been over layer height for percentage values there.  The tool tip mentions it and its been that way ever since I can remember.  
I just checked a 2.3 install and its the same in that too, % is over layer height.  You aren't getting confused wit Super Slicer are you ?  as Merill did change that to use nozzle size rather than layer height.

2Balu2, 2022-10-09:
> Its always been over layer height for percentage values there. The tool tip mentions it and its been that way ever since I can remember.
> I just checked a 2.3 install and its the same in that too, % [...]

Oh, yes that could be possible. I'm mainly using SuperSlicer. Hm, guess it can be closed if intended (it does make more sense to me to use the nozzle size, but it's not my decision)

rtyr, 2022-10-09:
It was always like that. We will likely change it, even though it will brake existing custom print profiles.

I am closing it as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2829.

closed 2022-10-09 by rtyr