prusa3d/PrusaSlicer #9023: PrusaSlicer Changes Scale without User Input

CamStLouis, 2022-10-09:
### Description of the bug

I have found the perfect oversize settings for my brand of ASA; which is 100.7% X/Y and 100.5% Z. However, for some objects, when I try to enter this into the scale field, the numbers change immediately after I hit enter. There is no current way to set objects to the correct scale when this happens. It can happen even when another object on the same print works fine. It is not fixed by saving and reloading the file.

[image]

<iframe width="640" height="400" src="https://screencast.autodesk.com/Embed/Timeline/37c8121a-267f-4fc1-ab2c-4cb0d1341387" frameborder="0" allowfullscreen webkitallowfullscreen></iframe>

### Project file & How to reproduce

[ScaleBug.zip](https://github.com/prusa3d/PrusaSlicer/files/9742012/ScaleBug.zip)

How to reproduce: try to change the scaling on both objects. One will fail.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10

### Printer model

Prusa Mini+

kubispe1, 2022-10-11:
Hi thank for report, I can confirm this issue (I repoduce it). Quite confusing issue. When I reset rotation, rotation works fine :D
[image]

Also when I export this object to stl and then import, scale works ok..

kubispe1, 2022-10-12:
This issue should be fixed with new release. 2.6.0. I don't know exactly what is a source of the problem, but I tested it and it looks OK. I am closing it at the moment.

closed 2022-10-12 by kubispe1