prusa3d/PrusaSlicer #9024: `wxTranslations::Get()->SetLanguage(...);` crashes

bernhardu, 2022-10-09:
### Description of the bug

I was trying to have a look at the crash reported to the Debian package in [this bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1020702).

It ends currently with this backtrace:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f2589f565f3 in std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_data (this=<optimized out>) at /build/gcc-12-GwPmq4/gcc-12-12.2.0/build/x86_64-linux-gnu/libstdc++-v3/include/bits/basic_string.h:234
234     /build/gcc-12-GwPmq4/gcc-12-12.2.0/build/x86_64-linux-gnu/libstdc++-v3/include/bits/basic_string.h: Datei oder Verzeichnis nicht gefunden.
(gdb) bt
#0  0x00007f2589f565f3 in std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_data (this=<optimized out>) at /build/gcc-12-GwPmq4/gcc-12-12.2.0/build/x86_64-linux-gnu/libstdc++-v3/include/bits/basic_string.h:234
#1  std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_is_local (this=<optimized out>) at /build/gcc-12-GwPmq4/gcc-12-12.2.0/build/x86_64-linux-gnu/libstdc++-v3/include/bits/basic_string.h:274
#2  std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::capacity (this=<optimized out>) at /build/gcc-12-GwPmq4/gcc-12-12.2.0/build/x86_64-linux-gnu/libstdc++-v3/include/bits/basic_string.h:1134
#3  std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_assign (this=this@entry=0x0, __str=L"") at /build/gcc-12-GwPmq4/gcc-12-12.2.0/build/x86_64-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:279
#4  0x00007f258896e05a in std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::assign (__str=L"", this=0x0) at /usr/include/c++/12/bits/basic_string.h:1540
#5  0x00007f258896e0c6 in wxTranslations::SetLanguage (this=0x0, lang=lang@entry=wxLANGUAGE_DEFAULT) at ./src/common/translation.cpp:1384
#6  0x0000563e57fb8e26 in Slic3r::GUI::GUI_App::load_language (this=0x563e59d29200, language=..., initial=<optimized out>) at ./src/slic3r/GUI/GUI_App.cpp:2079
#7  0x0000563e57fbaaf5 in Slic3r::GUI::GUI_App::on_init_inner (this=0x563e59d29200) at ./src/slic3r/GUI/GUI_App.cpp:1093
#8  0x0000563e57fbd4c6 in Slic3r::GUI::GUI_App::OnInit (this=<optimized out>) at ./src/slic3r/GUI/GUI_App.cpp:1035
#9  0x00007f258891feda in wxEntry (argc=<optimized out>, argv=<optimized out>) at ./src/common/init.cpp:487
#10 0x0000563e57f9f7b9 in Slic3r::GUI::GUI_Run (params=...) at ./src/slic3r/GUI/GUI_Init.cpp:54
#11 0x0000563e577605de in Slic3r::CLI::run (this=<optimized out>, argc=<optimized out>, argv=<optimized out>) at ./src/PrusaSlicer.cpp:618
#12 0x0000563e57735bf4 in main (argc=<optimized out>, argv=<optimized out>) at ./src/PrusaSlicer.cpp:844
```

Therefore the reason seems to be the method `SetLanguage` is called on a null pointer.
And this nullpointer is returned from following `wxTranslations::Get()` call.

https://github.com/prusa3d/PrusaSlicer/blob/c3bdf2fc28bec3c1af816447c3a2b056009cfefe/src/slic3r/GUI/GUI_App.cpp#L2090-L2097

I just wanted to ask if it would be a good idea to change this line to be able to handle this nullptr,
especially because wxwidget documentation explicitly notes `Returns current translations object, may return NULL.`
https://docs.wxwidgets.org/3.0/classwx_translations.html#ab384ea68c44e74cfd2cd59e782529540

That way probably it might be possible to startup, even if translations are not correct?

There is at least this [issue](https://github.com/prusa3d/PrusaSlicer/issues/8299#issuecomment-1186748653) which shows also this situation in a crash.

### Project file & How to reproduce

-

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+dfsg-1

### Operating system

Debian Bookworm/unstable

### Printer model

-

hyperair, 2022-10-11:
Looks like this was answered in https://github.com/prusa3d/PrusaSlicer/issues/8299#issuecomment-1236874810, I'll be applying the patch in Debian

bernhardu, 2022-10-21:
> Looks like this was answered in [#8299 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/8299#issuecomment-1236874810), I'll be applying the patch in Debian

That proves I did not catch the whole meaning of this comment when I first read it.
So I am going to close this issue, while it still could be considered to mark this source code location
with a comment that it is really relied on returning `wxTranslations::Get()` returning here a non-null value.

closed 2022-10-21 by bernhardu