prusa3d/PrusaSlicer #9028: Vase mode creates additional perimeters around thin gaps

MathieuGuennaoui, 2022-10-10:
### Description of the bug

When adding infolding features to my design, and trying different gap thicknesses to fine-tune the near crossing of the perimeters, the slicer couldn't handle 0,1mm gaps in vase mode, creating new perimeters instead. Moving to 0,11mm was enough to bring back the expected behavior. 

Here is a picture of the 0,1mm gap. The new perimeters are visible, instead of an invagination along the corners. 

[image]

### Project file & How to reproduce

The project contains the artefact around layer 130 (about 26mm of height). Resize the model at 101% to see the expected behavior, with normal perimeters. 

[PS_bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9748289/PS_bug.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+MacOS-arm64

### Operating system

MacOS Monterey 12.6

### Printer model

Prusa i3 MK3S MMU2S

DragonFinlay, 2022-10-12:
I can reproduce it.
1. There is a parameter to adjust this, it's the "Slice gap closing radius" in the advanced print settings.
The default value is 0.049mm, so it separates gaps of 0.098mm or more.
2. This should include your gap of 0.1mm, but I double-checked by measuring and saw that at the problematic heights the gap is closer to 0.097mm, which then creates these connecting walls.

So either increase the gap when modeling the part or decrease the Slice gap closing radius (which is what I did below), and you should be successful.

[image]

MathieuGuennaoui, 2022-10-12:
Thank you for your reply.

I didn't think of the interaction between the closing radius option and the inconsistency of the gap. I designed it on Fusion 360 though.

Could you please tell me how you managed to measure it ? The fillet prevents me from selecting both faces separately to get the dimensions.

DragonFinlay, 2022-10-13:
I used Meshmixer to measure it. Even though it's quite outdated it's still good for a handful of things like measuring 3D models. :)

Closing this as solved.