prusa3d/PrusaSlicer #9032: Please allow application of Variable Layer Height function at user specified height range

laptop-laptop, 2022-10-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

Using Modifiers, we could change some parameters within a specific range of heights set by the users. 
However, there is no Variable Layer Height function yet.

After clicking the Variable Layer Height icon at the top, we can only change the values of Adaptive and Smooth.
It applies the settings to the entire model. Could you please allow the user to specify the range of layer heights
at which the Variable Layer Height function is applied? 

**Describe how it would work**
See above

**Describe alternatives you've considered**
See above

**Additional context**

Prusa Support said I can only use a mouse to change the layer height (which is not precise nor repeatable) or use
the Variable Layer Height function which applies the changes to the entire model. Is this really true?

foreachthing, 2022-10-11:
Yeah, in Slic3r, there is that function called "Layer Height Table" (sort of a modifier):
[image]
**This is per model!!**

This function has been missing for a long time (I've been missing it since I switched to PS):
https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/layer-height-table-in-prusa-slicer/

What I do (as a work-around), is changing the printer settings (min/max layer height) to "paint" more precise layer heights. Annoying, but works.

If you really need control, you can unzip the .3mf  and edit the .\Metadata\Slic3r_PE_layer_heights_profile.txt file. First you'll need to change something and save your project as .3mf.
But it's a bit a try and error approach.

laptop-laptop, 2022-10-11:
Thanks for the suggestion. Could you please explain more on the procedure for the work-around? By changing the printer settings (min/max layer height), do you mean the values under Min and Max in Printer Settings->'Layer height limits'? How does that allow me to apply the settings in Variable Layer Height function to a specific range of layers?

kubispe1, 2022-10-12:
Hi, I quickly check out your comments and I think you are looking for this function. There you set height ranges similarly like in Slic3r.
[image]

foreachthing, 2022-10-12:
@kubispe1, dang!
I've never noticed that modifier before! That's what I've been missing!

And, as an added Bonus, you can add an Infill setting (to the height range) to combine every n layer. Example: if you use 0.05 mm layer thickness, you could add a "Combine infill every n layer" and set it to 4 => 0.2 mm thick infill.

DragonFinlay, 2022-10-12:
Closing as solved

closed 2022-10-12 by DragonFinlay

laptop-laptop, 2022-10-12:
Hi @kubispe1 I have been using your approach to change some parameters in specific layers. However, I cannot find the option to adjust the Adaptive and Smooth variables in Variable layer height menu. Clicking the cogwheel shows Infill, Layers and Parameters as as well as Add Settings. Under Add Settings, there are Advanced, Extrusion Width, Fuzzy Skin, Infill, Ironing, Layers and Perimeters and Speed. There is nothing on Adaptive and Smooth. Could you please let me know how to do, say Adaptive 1, Smooth 1 and click Smooth once at layers heights from 15.36 mm to 22.4 mm?

laptop-laptop, 2022-10-12:
> Hi, I quickly check out your comments and I think you are looking for this function. There you set height ranges similarly like in Slic3r. [image]

Hi @kubispe1 I have been using your approach to change some parameters in specific layers. However, I cannot find the option to adjust the Adaptive and Smooth variables in Variable layer height menu. Clicking the cogwheel shows Infill, Layers and Parameters as as well as Add Settings. Under Add Settings, there are Advanced, Extrusion Width, Fuzzy Skin, Infill, Ironing, Layers and Perimeters and Speed. There is nothing on Adaptive and Smooth. Could you please let me know how to do, say Adaptive 1, Smooth 1 and click Smooth once at layers heights from 15.36 mm to 22.4 mm?

kubispe1, 2022-10-13:
@laptop-laptop, there is only Variable layer height for this purpose. You can do it only manually there. Can you specify, what parameter should be in this input table? But, I afraid there is no implementation of this kind in near future...

laptop-laptop, 2022-10-13:
@kubispe1 Not in input table at the bottom right of your screenshot but if we could change the Adaptive, Smooth and number of smooth clicks parameters in the cogwheel above the table, that would be great!