prusa3d/PrusaSlicer #9033: Automatic adjustment of Color Change after application of Variable Layering Height function

laptop-laptop, 2022-10-11:
**Is your feature request related to a problem? Please describe.**

After using the Variable Layering Height function, I manually adjusted the Color Change levels to, say 1.75, 1.82 and 8.23 as the original heights of interested got changed due to application of the Variable Layering Height function. From the Gcode, I found that the slicer put M600 at the beginning of layer heights 1.8147, 1.85939 and 8.43365 rather than 1.75, 1.82 and 8.23. What is wrong here?

I went back to Variable Layering function and hit the Reset button.  This time the slicer automatically changed the Color Change levels to 1.8, 1.85 and 8.25. I checked the Gcode and M600 was placed at these layers correctly. So if Variable Layering Height function is used, layer heights set by Color Change is not automatically adjusted?

**Describe the solution you'd like**
Have the slicer to automatically adjusts the Color Change levels after application of Variable Layering Heigh function.

**Additional context**
Spent 2.5 hours with three Prusa support agents on various PS related issues. The first guy transferred me to another colleague whom he said knows more about PS. After some chatting, the second guy transferred me to the third guy who could not answer any questions.

laptop-laptop, 2022-10-12:
When Variable Layer Height is turned off, everything works fine. When it is turned on and I set Adaptive and Smooth values followed by setting of Color Change in the vertical bar by clicking the + sign, there is something wrong.

In 'After Setting Variable Layer Color change at 1.75.png", you can see that I set Color Change at 1.75mm.
In 'After Setting Variable Layer Color change at 1.82.png", you can see that I set Color Change at 1.82mm.

So, if we look at the gcode, there should be a M600 when the layer change occurs at 1.75 and 1.81mm.
However, from 'First Occurrence of M600.png' and 'Second Occurrence of M600.png', you can see that 
the first M600 is put during layer change to 1.8148mm while the second M600 is put during layer change
to 1.85939. They should be at 1.75 and 1.81 respectively.

From 'No M600 at layer 1dot7448.png', you see that there is no M600.
Similarly, from 'No M600 at layer 1dot7548.png', you see that there is also no M600.
So it seems that the issue may not be due to rounding error.

In this example, it seems that the first occurrence of M600 at 1.75 might have been skipped by the slicer. 

It seems that when Variable Layer Height is used, M600 were put in the wrong locations within the gcode either because automatic adjustment was not implemented or there is a bug somewhere. Once I deactivated Variable Layer Height by pressing the reset button, the M600 occurrences in the gcode were placed correctly. Tried PS 2.5.0 beta1 and the final 2.5.0 versions. Same problem. 

[image]

[image]

[image]

[image]

[image]

[image]

DragonFinlay, 2022-10-13:
Okay, I see now what you mean with the g-code, but I still can't reproduce it. I followed your steps:
Enable variable layer height -> Use adaptive/smooth settings -> Add color changes -> Slice and check G-Code

Here are my results:

[image]
[image]

[image]
[image]

Please try slicing my file, both unchanged and changing the variable layer height, and let me know your results. Thanks!
[VLH Test.zip](https://github.com/prusa3d/PrusaSlicer/files/9776019/VLH.Test.zip)

laptop-laptop, 2022-10-13:
I am very sorry. The situation actually is: Add color changes->Enable variable layer height->Apply changes (with adaptive/smooth settings under the menu)->Notice that layers where I set the color changes no longer correspond to the ones of interests due to the variable layer height changes->Readjust color changes->Slice and check G-Code. Should the end result be the same as the one you got?

It seems to depend on luck. Sometimes after variable layer height changes I don't need to readjust the layer heights in color change. However, sometimes I do need to readjust the layer heights and reapply color change.

I tried SuperSlicer 2.4.58.5. It is able to generate the gcode correctly. However, it has new issue for me. The hole sizes I created fit the screws well when sliced using Prusa Slicer. Under SuperSlicer, the holes no longer fit. Is there a simple way to fix this, perhaps with a parameter value change?

laptop-laptop, 2022-10-13:
What values did you use for Adaptive and Smooth? How many times did you press Smooth? I followed your steps but I could only put a Color Change marker at 5.95 (24), 6.20, 6.45. There is no 6.21.

Also, when I went to variable layer height, I saw that you have a very nice vertical variation bar. Only the top, say 10%, are in green while the rest are in orange with dark brown lines. How did you specify the range of layer heights where to apply the changes?

DragonFinlay, 2022-10-14:
No worries! I could reproduce it, and I think what's happening is that PrusaSlicer is always rounding up to the next highest layer.
I just tested it, if you program a layer change at e.g. 10.00 and after editing VLH you have one layer at 9.99 and one at 10.29, PrusaSlicer will shift the color change up those 0.29mm to 10.29 rather than down the 0.01mm to 9.99.

I'll check if we can fix this to round to nearest instead of next-highest. Thanks for helping me understand what's happening!

DragonFinlay, 2022-10-14:
Oh, and about your questions:
1. Have you tried XY size compensation to fix the size of the holes?

[image]
[image]

2. What settings I use for VLH depends on the model. Usually I go with 5 for adaptive and then I press smooth maybe 10 times to get a nice and even curve. :)

laptop-laptop, 2022-10-30:
> I'll check if we can fix this to round to nearest instead of next-highest. Thanks for helping me understand what's happening!

Hello, what is the status? 

As for XY Size Compensation, what values do you recommend

laptop-laptop, 2022-11-03:
It seems that there may be a bug in the G-code Viewer. 

In another project, after applying Variable Layer Height, I set color change at 6.23mm and 6.26mm. The generated Gcode has M600 at 6.23045 and 6.39751. I deleted the M600 at 6.39751 and put it at 6.26456 which is the AFTER_LAYER_CHANGE after 6.39751.

I loaded it to G-code Viewer. Strangely speaking, the markers for color change showed up at 6.23 and 6.40 (which I removed the M600) rather than 6.23 and 6.26. What is wrong here?

DragonFinlay, 2022-11-03:
Hi, sorry for only responding now.

We have it on the list and are checking how much it would take to make this change. I hope it doesn't cause too many problems in the meanwhile.

For your question about XY compensation, it really depends on the model. The value is the radius, so if you enter 0.05, the model will expand 0.1mm in diameter.

The "bug" in the g-code viewer sounds like it just takes the color change from something else, or "more" than just the M600. It's probably looking for the comment. I think that's just the nature of how the viewer works and changing the g-code yourself can confuse it.

laptop-laptop, 2022-11-06:
Thank you. 

While waiting for it to be changed, in the meantime what suggestion do you have? Do you think just moving the M600 manually to the correct layer (6.26456 in the above simple example) would work even the Preview shows otherwise? My recent prints each takes about a day to complete. I can't afford to try different methods out especially the printer is in my small bedroom.

DragonFinlay, 2022-11-08:
If you change the g-code correctly (M600 is the definitive command for the printer to pause for filament change, so moving it to the correct position will work out. The printer doesn't care about comments.

closed 2026-09-29 by prusaslicer-bot[bot]