prusa3d/PrusaSlicer #9038: Allow Sub-Features in Profiles

3DCoded, 2022-10-13:
**Is your feature request related to a problem? Please describe.**
When using an E3D Revo Six on my Prusa, I like to add a line to the end G-Code to retract to help with nozzle swaps. I also added start G-Code to increase mesh bed leveling to 7x7. This all works, but it's tedious to add the settings to all my nozzle sixes and filaments, etc.

**Describe the solution you'd like**
 Is there a way to have "features" that will be part of profiles?

**Describe how it would work**
In the profile settings, there would be a features settings pane and you would select the features you want. You could add features from this pane too, similar to how the "Add Settings" on objects work, where you select the major setting group, and then check the ones you want to change.

**Describe alternatives you've considered**
I could manually do this on all my profiles, but that is very tedious.

**Additional context**
Maybe even have features in the configuration wizard so if someone has a Revo, they could just download the feature to retract in the end G-Code. It could also work if someone wants more dense mesh bed leveling, if they want to use a Volcano, Bondtech CHT, etc.

rtyr, 2022-10-13:
> When using an E3D Revo Six on my Prusa, I like to add a line to the end G-Code to retract to help with nozzle swaps. 

You can easily do that and save it as a new printer profile. Frankly I don't see anything tedious about it even if you have to do it for each nozzle diameter. You have custom HW and want to use custom start g-code and end g-code (for whatever reason), so making custom printer profile makes sense.

> I also added start G-Code to increase mesh bed leveling to 7x7.

There is no need for that, it can be easily configured directly on the printer side.

gudvinr, 2022-10-13:
> Frankly I don't see anything tedious about it even if you have to do it for each nozzle diameter.

There is an issue with that that rises from lack of proper inheritance model in user profiles. If you need to change start gcode or some extruder parameters like retraction settings, it gets very tedious once you have more than one.  
I believe that there are issues about exactly that.

Area5142, 2022-10-14:
Just a thought...

Maybe I'm thinking too much like a programmer, but an include directive everywhere g-code can be inserted would solve the problem of _"common"_ g-code.

Something like:
`$include <text_path_filename>`

For example:
`$include My_Printer_.6mm_Nozzle_Start_Code.gcode`

If only filename is specified, the file should be located with the profile, otherwise specify relative path to file (allows location on network drive).
When PrusaSlicer generates g-code, the contents are inserted where the include directive is.

This will make it easy to update specific setups for many profiles at once - just edit the common include file.
This is an advanced setup, as the common file has to be created and maintained outside PrusaSlicer.

closed 2026-09-29 by prusaslicer-bot[bot]