prusa3d/PrusaSlicer #9039: Add option change bed temp (actually turn off or lower) certain time before print end.

vovodroid, 2022-10-13:
It would be nice at the end of print to get bed cool enough to remove object immediately.

As alternative add placeholders for total print time/elapsed (fir instance in seconds) available on layer change. Thus we could add something like

if {total_print}-{elapsed} < 600 M140 S0

foreachthing, 2022-10-13:
Well, with a little math you can do that already:
[image]
Eye-ball the time (runtime in layer scrollbar) to turn the bed off, add custom g-code (left click where, then right click and add code) and voilà!

_But, I'd be afraid to lose my print just in the last second, because it doesn't stick anymore..._

vovodroid, 2022-10-15:
> with a little math you can do that already:

Sure, but it would be nice to get it in more handy form ))

> , I'd be afraid to lose my print just in the last second, because it doesn't stick anymore...

It depends on filament, bed, model bottom area... Sometimes it could stick too strong.

foreachthing, 2022-10-15:
> Sure, but it would be nice to get it in more handy form ))

Agreed 😄

l29ah, 2023-12-14:
In klipper you can use the following macro that utilizes prusaslicer's remaining time gcodes:
```
[gcode_macro M73] 
gcode: 
        {% if params.R == '1' %}        ; one minute before ending the print 
        M140 S0 ; turn off heated bed 
        {% endif %} 
        RESPOND TYPE=command MSG="{ params.R|int } mins | { params.P|int } %"
```

closed 2026-09-29 by prusaslicer-bot[bot]