prusa3d/PrusaSlicer #9040: PrusaSlicer crashes whenever Variable Layer Height button is clicked

jzobac, 2022-10-13:
### Description of the bug

Jakmile kliknu na variabilní výsku vrstvy, tak PrusaSlicer spadne. Mám jen integrovanou grafiku od Intelu.

PrusaSlicer Verze 2.5.0+linux-x64-GTK3; Ubuntu 22.04.1 LTS; x64; X11; Gnome 42.2

Díky, Jirka

### Project file & How to reproduce

Jakmile kliknu na variabilní výsku vrstvy, tak PrusaSlicer spadne. Mám jen integrovanou grafiku od Intelu.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+linux-x64-GTK3-202209060725.AppImage; PrusaSlicer-2.5.0+linux-x64-GTK2-202209060714.AppImage

### Operating system

Ubuntu 22.04.1; x64; X11

### Printer model

Mini+

kubispe1, 2022-10-14:
Ahoj Jirko, hodilo by se nám vědět další detaily, kdy se ti problém děje. Jinak nejsme schopní problém ohledat, reprodukovat a vyřešit.
Detaily typu: Je to u specifického modelu, nebo se ti to děje vždy ? (Případně prosím připoj .3mf).

jzobac, 2022-10-14:
Zdravim,
Děje se ti vždy, dokonce když si v Kubuntu (PC) naklikám projekt s variabilní výškou, tak tento projekt nejde ani otevřít v Ubuntu (ntb) aplikace vždy spadne.

J.

jzobac, 2022-10-14:
Log terminálu

./PrusaSlicer-2.5.0+linux-x64-GTK3-202209060725.AppImage -loglevel 5 > log.txt
*   Trying 185.115.1.124:443...
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* ALPN, offering http/1.1
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=files.prusa3d.com
*  start date: Sep  6 11:04:07 2022 GMT
*  expire date: Dec  5 11:04:06 2022 GMT
*  subjectAltName: host "files.prusa3d.com" matched cert's "files.prusa3d.com"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2 HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.5.0+linux-x64-GTK3
Accept: */*

* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: nginx
< Date: Fri, 14 Oct 2022 10:05:13 GMT
< Content-Type: text/html; charset=UTF-8
< Transfer-Encoding: chunked
< Connection: keep-alive
< Vary: Accept-Encoding
< Vary: Accept-Encoding
< 
* Connection #0 to host files.prusa3d.com left intact
*   Trying 185.115.1.124:443...
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* ALPN, offering http/1.1
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=files.prusa3d.com
*  start date: Sep  6 11:04:07 2022 GMT
*  expire date: Dec  5 11:04:06 2022 GMT
*  subjectAltName: host "files.prusa3d.com" matched cert's "files.prusa3d.com"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.5.0+linux-x64-GTK3
Accept: */*

* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: nginx
< Date: Fri, 14 Oct 2022 10:05:13 GMT
< Content-Type: text/html; charset=UTF-8
< Transfer-Encoding: chunked
< Connection: keep-alive
< Vary: Accept-Encoding
< Vary: Accept-Encoding
< 
* Connection #0 to host files.prusa3d.com left intact
*   Trying 185.115.1.124:443...
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* ALPN, offering http/1.1
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=files.prusa3d.com
*  start date: Sep  6 11:04:07 2022 GMT
*  expire date: Dec  5 11:04:06 2022 GMT
*  subjectAltName: host "files.prusa3d.com" matched cert's "files.prusa3d.com"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//1.5.2.ini HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.5.0+linux-x64-GTK3
Accept: */*

* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: nginx
< Date: Fri, 14 Oct 2022 10:05:13 GMT
< Content-Type: text/html; charset=UTF-8
< Transfer-Encoding: chunked
< Connection: keep-alive
< Vary: Accept-Encoding
< Vary: Accept-Encoding
< 
* Connection #0 to host files.prusa3d.com left intact
Neoprávněný přístup do paměti (SIGSEGV) (core dumped [obraz paměti uložen])
[log.txt](https://github.com/prusa3d/PrusaSlicer/files/9784851/log.txt)

Obraz paměti nevím kam se snažil uložit.

Sargastic, 2022-11-23:
Hello,

Same crashes here, as soon as I click on the 'Variable Layer Height' button.
Gentoo Linux, prusa slicer 2.5.0, Inter graphics card (on the mobo).
Compiled the software with debug, here is the backtrace when it crashes

(gdb) bt
#0  0x00007fffd34cf295 in  () at /usr/lib64/dri/crocus_dri.so
#1  0x00007fffd34d3f3a in  () at /usr/lib64/dri/crocus_dri.so
#2  0x00007fffd34d6958 in  () at /usr/lib64/dri/crocus_dri.so
#3  0x00007fffd350d835 in  () at /usr/lib64/dri/crocus_dri.so
#4  0x00007fffd389fd5b in  () at /usr/lib64/dri/crocus_dri.so
#5  0x00007fffd38a1d09 in  () at /usr/lib64/dri/crocus_dri.so
#6  0x00007fffd384e92d in  () at /usr/lib64/dri/crocus_dri.so
#7  0x00007fffd39b77f1 in  () at /usr/lib64/dri/crocus_dri.so
#8  0x00007fffd39baae1 in  () at /usr/lib64/dri/crocus_dri.so
#9  0x00007fffd3a7d8d6 in  () at /usr/lib64/dri/crocus_dri.so
#10 0x00007fffd360b97f in  () at /usr/lib64/dri/crocus_dri.so
#11 0x00007fffd3606702 in  () at /usr/lib64/dri/crocus_dri.so
#12 0x00007fffd3606feb in  () at /usr/lib64/dri/crocus_dri.so
#13 0x00007fffd3607221 in  () at /usr/lib64/dri/crocus_dri.so
#14 0x00007fffd33467fb in  () at /usr/lib64/dri/crocus_dri.so
#15 0x00007fffd33080b1 in  () at /usr/lib64/dri/crocus_dri.so
#16 0x00007fffd33083e2 in  () at /usr/lib64/dri/crocus_dri.so
#17 0x00005555562a2bda in Slic3r::GUI::GLCanvas3D::_mouse_to_3d(Slic3r::Point const&, float*) (this=this@entry=0x555559993fa0, mouse_pos=..., z=z@entry=0x0)
    at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:5732
#18 0x00005555562a8be9 in Slic3r::GUI::GLCanvas3D::render() (this=0x555559993fa0) at /usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:122
#19 0x00005555562a9210 in Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen() (this=0x555559993fa0) at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:4863
#20 Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen() (this=0x555559993fa0) at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:4855
#21 0x00005555562a9377 in Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) (evt=..., this=0x555559993fa0) at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:2265
#22 Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) (this=0x555559993fa0, evt=...) at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:2240
#23 0x00007ffff6f0fd8e in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#24 0x00007ffff6f1010a in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#25 0x00007ffff6f101a0 in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#26 0x00007ffff6f1025b in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#27 0x00007ffff6f10331 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#28 0x00007ffff6f10087 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#29 0x00007ffff6b2d05f in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#30 0x00007ffff6b2d038 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#31 0x00007ffff6b2d038 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#32 0x00007ffff6b2d038 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#33 0x00007ffff69ce3af in wxFrame::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#34 0x00007ffff6a1115d in wxAppBase::ProcessIdle() () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#35 0x00007ffff6941115 in wxApp::DoIdle() () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#36 0x00007ffff6941223 in  () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#37 0x00007ffff5a177c0 in g_main_context_dispatch () at /usr/lib64/libglib-2.0.so.0
#38 0x00007ffff5a17b68 in  () at /usr/lib64/libglib-2.0.so.0
#39 0x00007ffff5a17e53 in g_main_loop_run () at /usr/lib64/libglib-2.0.so.0
#40 0x00007ffff5da4bcd in gtk_main () at /usr/lib64/libgtk-3.so.0
#41 0x00007ffff695c635 in wxGUIEventLoop::DoRun() () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#42 0x00007ffff6de992d in wxEventLoopBase::Run() () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#43 0x00007ffff6db0c45 in wxAppConsoleBase::MainLoop() () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#44 0x00007ffff6e2eec8 in wxEntry(int&, wchar_t**) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#45 0x0000555555f557d6 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GUI_Init.cpp:54
#46 0x0000555555805fb0 in Slic3r::CLI::run(int, char**) (this=0x7fffffffd7f0, argc=1, argv=0x7fffffffda28) at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/PrusaSlicer.cpp:618
#47 0x00005555557ec191 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/PrusaSlicer.cpp:844

If you need any more data, or testing patches, just ask.

Sargastic, 2022-11-23:
If using a model that already has variable layer height (https://www.printables.com/model/310661-cocoon-remote-stand-two-sizes), the slicer crashes if we cick on the 'variable layer height' button in the right part of the screen.

Backtrace : 

Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x00007fffd34cf295 in ?? () from /usr/lib64/dri/crocus_dri.so
(gdb) bt
#0  0x00007fffd34cf295 in  () at /usr/lib64/dri/crocus_dri.so
#1  0x00007fffd34d3f3a in  () at /usr/lib64/dri/crocus_dri.so
#2  0x00007fffd34d6958 in  () at /usr/lib64/dri/crocus_dri.so
#3  0x00007fffd350d835 in  () at /usr/lib64/dri/crocus_dri.so
#4  0x00007fffd389fd5b in  () at /usr/lib64/dri/crocus_dri.so
#5  0x00007fffd38a1d09 in  () at /usr/lib64/dri/crocus_dri.so
#6  0x00007fffd384e92d in  () at /usr/lib64/dri/crocus_dri.so
#7  0x00007fffd39b77f1 in  () at /usr/lib64/dri/crocus_dri.so
#8  0x00007fffd39baae1 in  () at /usr/lib64/dri/crocus_dri.so
#9  0x00007fffd3a7d8d6 in  () at /usr/lib64/dri/crocus_dri.so
#10 0x00007fffd360b97f in  () at /usr/lib64/dri/crocus_dri.so
#11 0x00007fffd3606702 in  () at /usr/lib64/dri/crocus_dri.so
#12 0x00007fffd3606feb in  () at /usr/lib64/dri/crocus_dri.so
#13 0x00007fffd360ce78 in  () at /usr/lib64/dri/crocus_dri.so
#14 0x00007fffd31af0f5 in  () at /usr/lib64/dri/crocus_dri.so
#15 0x00007fffd30c6fb1 in  () at /usr/lib64/dri/crocus_dri.so
#16 0x00007ffff03ccd9d in glPrimitiveBoundingBox () at /usr/lib64/libGLX_mesa.so.0
#17 0x00007ffff03bfe31 in  () at /usr/lib64/libGLX_mesa.so.0
#18 0x00007ffff03af12b in  () at /usr/lib64/libGLX_mesa.so.0
#19 0x00007ffff63fdbfb in wxGLCanvasX11::SwapBuffers() () at /usr/lib64/libwx_gtk3u_gl-3.0-gtk3.so.0
#20 0x00005555562a8d38 in Slic3r::GUI::GLCanvas3D::render() (this=0x555559994200)
    at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:1556
#21 0x00005555562a9210 in Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen() (this=0x555559994200)
    at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:4863
#22 Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen() (this=0x555559994200)
    at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:4855
#23 0x00005555562a9377 in Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) (evt=..., this=0x555559994200)
    at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:2265
#24 Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) (this=0x555559994200, evt=...)
    at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GLCanvas3D.cpp:2240
#25 0x00007ffff6f0fd8e in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#26 0x00007ffff6f1010a in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#27 0x00007ffff6f101a0 in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#28 0x00007ffff6f1025b in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#29 0x00007ffff6f10331 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#30 0x00007ffff6f10087 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#31 0x00007ffff6b2d05f in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#32 0x00007ffff6b2d038 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#33 0x00007ffff6b2d038 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#34 0x00007ffff6b2d038 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#35 0x00007ffff69ce3af in wxFrame::SendIdleEvents(wxIdleEvent&) () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#36 0x00007ffff6a1115d in wxAppBase::ProcessIdle() () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#37 0x00007ffff6941115 in wxApp::DoIdle() () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#38 0x00007ffff6941223 in  () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#39 0x00007ffff5a177c0 in g_main_context_dispatch () at /usr/lib64/libglib-2.0.so.0
#40 0x00007ffff5a17b68 in  () at /usr/lib64/libglib-2.0.so.0
#41 0x00007ffff5a17e53 in g_main_loop_run () at /usr/lib64/libglib-2.0.so.0
#42 0x00007ffff5da4bcd in gtk_main () at /usr/lib64/libgtk-3.so.0
#43 0x00007ffff695c635 in wxGUIEventLoop::DoRun() () at /usr/lib64/libwx_gtk3u_core-3.0-gtk3.so.0
#44 0x00007ffff6de992d in wxEventLoopBase::Run() () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#45 0x00007ffff6db0c45 in wxAppConsoleBase::MainLoop() () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#46 0x00007ffff6e2eec8 in wxEntry(int&, wchar_t**) () at /usr/lib64/libwx_baseu-3.0-gtk3.so.0
#47 0x0000555555f557d6 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...)
    at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/slic3r/GUI/GUI_Init.cpp:54
#48 0x0000555555805fb0 in Slic3r::CLI::run(int, char**) (this=0x7fffffffd7a0, argc=1, argv=0x7fffffffd9d8)
    at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/PrusaSlicer.cpp:618
#49 0x00005555557ec191 in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /usr/src/debug/media-gfx/prusaslicer-2.5.0/PrusaSlicer-version_2.5.0/src/PrusaSlicer.cpp:844

bubnikv, 2022-11-24:
> in main(int, char**) (argc=, argv=)

This Linux specific crash will be fixed in PrusaSlicer 2.6.0.

jzobac, 2022-11-24:
Je predbezne mozne rici termin vydani? Nejsem si jisty jestli je problem v Ubuntu a nebo v tom ze Ubuntu mam posledni verzi 22.04. Kazdopadne radeji drzim stolni PC na Kubuntu 20.04 a sluzebni ntb take na Kubuntu 20.04. Jeden stroj vzdy aktualizuji jako prvni a zbyle az s nejakym odstupem.

bubnikv, 2022-11-24:
Problém je v OpenGL, 2.6 používá OpenGL jinak.

kubispe1, 2023-02-20:
@jzobac ahoj, mohl bys otestovat s novou alfou? Věřím, že by problém mohl být vyřešen, ale raději prosím o test.
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha4

jzobac, 2023-03-26:
Zdravim,
potvrzuji že již nepadá.

Díky J.

lukasmatena, 2023-03-26:
Diky za info.

closed 2023-03-26 by lukasmatena