prusa3d/PrusaSlicer #9042: Null info on loading

welbot, 2022-10-13:
### Description of the bug

Just had a user come to JoelTelling's discord saying they couldn't load PrusaSlicer 2.5 and got a message about null info.
I helpled them troubleshoot, and it was a corrupt PrusaSlicer.ini file full of zero's as a result of having turned the printer off and on again.
Got them to delete the ini and reload and all was well again. 
Reason the solution jumped to mind, was I recall seeing something about this issue supposedly being dealt with in the 2.5 alpha, but clearly the fix didn't work.

### Project file & How to reproduce

No project file as Slicer was unable to be opened. Didn't get screenshot of the bug as it was from another user, but the info from his level 5 logging was as follows:

C:\Program Files\Prusa3D\PrusaSlicer>prusa-slicer-console.exe --loglevel=5
[2022-10-13 14:23:58.820467] [0x00001d18] [trace]   Initializing StaticPrintConfigs
[2022-10-13 14:24:00.150327] [0x00001d18] [info]    The configuration file C:\Users\tyric\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini has a wrong MD5 checksum or the checksum is missing. This may indicate a file corruption or a harmless user edit.
[2022-10-13 14:24:00.151320] [0x00001d18] [info]    The configuration file C:\Users\tyric\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini is corrupted, because it is contains null characters.
[2022-10-13 14:24:00.282626] [0x00001d18] [info]    Failed to parse configuration file "C:\Users\tyric\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini": The configuration file contains null characters.
Error parsing PrusaSlicer config file, it is probably corrupted. Try to manually delete the file to recover from the error. Your user profiles will not be affected.

C:\Users\tyric\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini

The configuration file contains null characters.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Prusa MK3

stepikovo, 2022-10-19:
Are you sure it happened before the user turned on the printer? How is the printer related to an ini file?

welbot, 2022-10-20:
I can only go off what they told me. They were printing something, and had to turn it off for some reason mid print, and then after that, the slicer wouldn't open.
While I was helping them to troubleshoot, I came across a closed issue on here, that stated this was a known occasional issue, and had been addressed in 2.5a. I can't find the post now, but it was for someone using 2.4.x. Not sure of exact one though.

pokemane, 2023-01-08:
This just happened for me too, though I don't have a printer connected to PrusaSlicer. I just haven't used it in a couple of days. It may have shut down or may have been exited as a result of some other application/installer needing to restart the computer, I am not sure. Exact same error message. Using 2.5.0, Windows 11.

numo68, 2023-02-10:
Same completely reproducibly happened here when the (virtual) machine I am slicing on crashed due to some VMware Workstation 3D graphics acceleration woes. My guess is that the PrusaSlicer keeps the file open for writing instead of closing (or at least flushing/sync-ing) it when done with the write.

My build: 2.5.0+win64-202209060714 on Win 10, pretty stock config.

bubnikv, 2023-02-10:
Also in Bambula
https://twitter.com/stlDenise3D/status/1623878580814901249
https://twitter.com/bubnikv/status/1623969126967394306

bubnikv, 2023-02-10:
> My guess is that the PrusaSlicer keeps the file open for writing instead of closing (or at least flushing/sync-ing) it when done with the write.

Frankly it seems to be a hard drive / windows issue not flushing files reliably if they are saved when shutting down Windows.

numo68, 2023-02-10:
> Frankly it seems to be a hard drive / windows issue not flushing files reliably if they are saved when shutting down Windows.

Yeah, also a possibility. Would be nice to know whether that always happened when crashing / shutting down the OS. On a regular shutdown/reboot the files should be closed normally, also the application has a way to be notified of those events with a time to react.

I was not able to reproduce that with force-killing the slicer but not sure whether windows has an equivalent of "kill -9".

numo68, 2023-02-10:
> It happens on Windows only and it is extremely, really extremely rare. We went to such lengths as to save the file twice, adding checksums etc, but there seems to be an issue with Windows and we cannot do much about that.

I can understand the head scratching - being a developer myself - but I had my amount of windows crashes and it is very very unusual to see a file corrupted in a way that there is about 600 bytes written fine, the rest is zeros and the size is the "correct" one. I can also understand that if the frequency is low enough and never happens when the exe is closed regularly to just ignore.

However, the slicer for example rewrites the file immediately after it is started. The one and only change is literally the date at the top and checksum at the bottom. In my case the vmware caused crash happened right after, so it reproducibly hit the vulnerable period, but the whole write could be probably deferred to when something important actually changes.

numo68, 2023-02-10:
fwiw, just tried restarting Windows when the slicer was running. This did not reproduce the problem but I got

Application popup: prusa-slicer.exe - Application Error : The instruction at 0x00007FFDA4F6E897 referenced memory at 0x000002A6B9A15E90. The memory could not be read.

3dDesigner360, 2024-04-15:
[image]

3dDesigner360, 2024-04-15:
downloaded 5 times same, found the last 4 updated files tried them too same result. Turned it off and back on again ...... Im at a loss because it worked fine one day and then this. would love a solution. also changed locations on hard drive and ssd and no matter where i put it and how much i deleted and re installed it still wont correct

neophyl, 2024-04-15:
3dDesigner, your issue is not related to the original ones.  It sounds like you have a corrupted config file (as the error message tells you).  Config files are not removed on uninstall and the location doesn't change even if you install to a different location.
You can rename or delete the config folder and then you should be able to reinstall and start PS.
The other option is to go to the listed location of the ini file and try and find the erroneous characters to manually repair your .ini file.  That option though needs a bit of experience to figure out what is wrong.

numo68, 2024-12-24:
another instance seen on facebook user group, with the user wondering what the ... has this to do with a roaming, he is running on a PC and not phone :-D Also he was not able to find the folder, as the AppData is hidden.

Would it be perhaps possible to ask for a confirmation, delete the file and try again, instead of sending the user to the parts of Windows he/she does not know nor care about? It is obviously not salvageable and doing it would make the UX better...

closed 2026-09-29 by prusaslicer-bot[bot]