prusa3d/PrusaSlicer #9044: Auto speed printing slow for no apparent reason

pylorak, 2022-10-14:
### Description of the bug

Using Auto Speed causes low print speeds without obvious reason. In the attached project, the maximum volumetric flow rate is never higher than 3 mm2/s (instead of 10), and the printer head speed is almost always below 30mm/s (instead of 50). Only an extremely few and short segments are generated at higher speeds, in weird places like a tight 180°turn, tiny gap fills or when two extrusions touch.

Relevant settings in the project:
Max volumetric flow rate in Filament settings: 10 mm2/s
Auto speed volumetric flow rate: 10 mm2/s
Auto speed print speed: 50 mm/s
Auto cooling: Disabled
Speed settings: Set to Auto (0) where accepted, support 50, bridges 40, gap fill 20.

[image]
[image]
[image]

### Project file & How to reproduce

[slow_print_report.zip](https://github.com/prusa3d/PrusaSlicer/files/9786842/slow_print_report.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Voxelab Aquila

cimbalek, 2022-10-19:
Same problem, hint for `max print speed` also says  "when setting other speed seetings to 0 PS will autocalculate the optimal speed..." I have no speed settings to 0 and it still tries to autocalculate. Quality of print went up with new PS, but sometimes I prefer speed.

[image]

[image]

neophyl, 2022-10-19:
For cimbalek, check your settings under filament, slow down if layer time is below xx.  Those often cause slower layers on smaller objects just like you have pictured.

pylorak, 2022-10-19:
@Tupson444, based on your description, your issue is that Auto Speed is calculating the speeds inconsistently for different layers.

@cimbalek You are describing that Auto Speed is influencing speed calculations even though no specific speed setting is set to use this feature.

I reported a different problem here. This ticket is about Auto Speed calculating too slow speeds, even though it could print faster because no other setting is limiting it in material flow, printer speed, or layer time. While I cannot know or say whether your issues are truly unrelated, on the surface at least they are very different phenomenons. So I think both of you should open separate tickets about your problems.

lvitol, 2022-10-25:
Same problem, Auto Speed calculating too slow speeds when using arachne perimeter generator. Auto Speed works as expected if use old perimeter generator.

cimbalek, 2022-10-25:
@Tupson444 please disregard. I'll need to revisit this problem because I found that `slow down if layer print time is below` had a high value set and was influencing that. I'm sorry that I missed it.

pylorak, 2022-10-25:
> Same problem, Auto Speed calculating too slow speeds when using arachne perimeter generator. Auto Speed works as expected if use old perimeter generator.

I cannot confirm this. Using the same project that is attached to this issue in the top post, using the classic engine is even much worse.

With Arachne:
[image]

With Classic:
[image]

Tupson444, 2022-11-01:
@pylorak Yeah, you are right, it's probably unrelated. I thought the issues might share the same cause (who knows?).

@lvitol It seems that my issue is also related to Arachne. I've opened a new thread: #9112

pylorak, 2023-02-19:
Tested today with PrusaSlicer 2.6.0-alpha4 to see if maybe solved. Issue still exists in new version too.

LightningShark25, 2023-12-29:
I have noticed this on previous versions of prusaslicer and also on the latest version 2.7.x

Auto Cooling on or off doesn't make a difference
Confirmed that the Filament Max flow is what I expect
The max flow rate is in excess of the filament.
Setting the speed to say max 100mm/s I get some random low speed.  If I set the max speed to 500, then the speed is limited by the filament flow rate - but this isn't super helpful as I don't want some features printed at the calculated 280mm/s  

Setting has great potential - but just doesn't work

BloodyRain2k, 2023-12-30:
I've also ran into this problem in the past, but so far it wasn't too extreme. But now I found an extreme example:

[image]

I don't know why, but it's auto calculating the slowest print speed **below** the minimum print speed set for the printer (8mm/s).
And then also decides to use that excessively slow speed for the **entire** model...

As soon as I prevent it from auto calculating the `Infill speed` everything seems fine:

[image]

I've also tried switching from Arachne to Classic, didn't change much besides raising the minimum speed from 1.1 to 4.1 mm/s.
It's also not the supports or `Fill gaps`. I'm also not sure how it manages to go *above* 60 mm/s when told to do 60 mm/s.

Here's the project file: [broken_auto_speed.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13798082/broken_auto_speed.3mf.zip)

TheMechaniac, 2024-01-20:
I've been dealing with the same bug, but found a work-around:

1. Set the "Speed for print moves" base settings (Perimeters, Infill) to the desired maximum auto speed (same speed as "Auto Speed (advanced) Max print speed")
2. Change "Speed for print moves" variant settings (small perimeters, external perimeters, solid infill, top solid infill, etc.) to 100% (only if auto speed is desired, leave unmodified if other speeds are required)
3. Print speed is now automatically limited by the lower of the two Max volumetric speed (print settings or filament settings), or the "Speed for print moves" base settings.

It appears as though "Auto Speed (advanced) Max print speed" does not interface with Auto (set to 0) speed settings in the way we think it should. Setting either max volumetric speed to 0 does appear to correctly surrender control, however.

BloodyRain2k, 2024-02-05:
> It appears as though "Auto Speed (advanced) Max print speed" does not interface with Auto (set to 0) speed settings in the way we think it should. Setting either max volumetric speed to 0 does appear to correctly surrender control, however.

The other thing about "Auto Speed" that's weird is that it's a global setting instead of a Printer setting. Maybe I understood something wrong about 3D printing, but so far I thought that the printer / hotend is what determines the max speed one should use.
So it's confusing that it's not tied to the printer's settings instead. There's probably cases where a profile with a lower setting could be useful, but in general I'd expect the printer to set the limit.

aknezevi-work, 2025-05-09:
I figured it is probably mostly due to Arachne and Gap Fill. It, therefore, heavily depends on geometry.
I am using Variable Layer Height feature, which is very powerful, and first layer height is kind of a variable layer height.
However, having set speeds hard coded as mm/s when using Variable Layer Height is bad because print will either go too fast on some places or too slow when it could be printing faster.
Therefore, I want to use auto speed.
It is just not consistent with speed giving. It will unnecessarily slow down the print speed sometimes, depends really on the Variable Layer Height and the amount of times i press "smooth".
I have several projects, sometimes Variable Layer Height "solves" the issue and the print is faster, sometimes it makes the issue and the print is slower. Sometimes it makes no difference. Depends on the geometry, when it really should not so much.

[Auto_Speed_3.zip](https://github.com/user-attachments/files/19976094/Auto_Speed_3.zip)
[Auto_Speed_4.zip](https://github.com/user-attachments/files/19976093/Auto_Speed_4.zip)

[Auto_Speed.zip](https://github.com/user-attachments/files/19976106/Auto_Speed.zip)

closed 2026-09-29 by prusaslicer-bot[bot]