prusa3d/PrusaSlicer #9046: Thumbnails ignore modifiers

Sargastic, 2022-10-15:
### Description of the bug

The generated thumbnaisl do not take modifiers (negative ones, at least) into account.

### Project file & How to reproduce

Here are a (useless) 3MF and associated GCODE, and scrennshots.
[Bug-01.zip](https://github.com/prusa3d/PrusaSlicer/files/9791740/Bug-01.zip)
[Screenshots.zip](https://github.com/prusa3d/PrusaSlicer/files/9791743/Screenshots.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Linux Gentoo

### Printer model

Prusa MK3 S

DragonFinlay, 2022-10-21:
The issue is that the thumbnail displays the unsliced 3D model. When using modifiers, the mesh is not actually changed, so the preview doesn't display modifiers.
I'm not sure if we will have thumbnail options in the future, but for now there's unfortunately not much you can do.

Keeping this open for future reference.

Sargastic, 2022-11-23:
Ok no big deal, thanks for the answer.

benjaminBoehme, 2026-09-28:
The issue still exists in PrusaSlicer 3.0.0(-alpha12), though it remains more of an inconvenience than a real problem. Maybe it'll still get addressed someday.
Interestingly, positive volume modifiers are rendered correctly, while negative modifiers are not.