prusa3d/PrusaSlicer #9049: Add ability to create !"printable toggle shortkey

wikolii, 2022-10-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

GAZ082, 2023-09-16:
I was about to open a ticket for this. Nice to have! I'd suggest the space bar.

GAZ082, 2023-09-16:
This issue is kinda old and shouldnt be too hard to implement, care to take a look  @lukasmatena ?

armorhide406, 2023-11-17:
So... has this been implemented or not?

closed 2026-09-29 by prusaslicer-bot[bot]