prusa3d/PrusaSlicer #9051: Multimaterial Painting bridges adds perimeters in thin air

vdubedout, 2022-10-16:
### Description of the bug

When using the multimaterial painting tool to paint bridges, the bridges can't hold. They drop because they are not extruded over a wall but side to a wall. 

[image]

Expectations: 
I would like that the painting on bridges works like a normal bridge by extending above a wall to let the bridge to correctly sit. 

### Project file & How to reproduce

[multimaterial_painting_1.zip](https://github.com/prusa3d/PrusaSlicer/files/9793416/multimaterial_painting_1.zip)

- Import STL
- Paint Bridge Wall
- Slice

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 

### Operating system

Win64 (win 10)

### Printer model

Voron Trident with multimaterial "single nozzle"

rtyr, 2022-10-16:
Enable "Interface shells" at the Print settings -> Multiple Extruders.

vdubedout, 2022-10-17:
[image]
[image]
[image]

@rtyr no changes unfortunately

rtyr, 2022-10-17:
I see, I thought there is some infill beneath it. This is known issue, and it is indeed not ideal behavior.

We already have issue on this topic https://github.com/prusa3d/PrusaSlicer/issues/6135. I am closing this one as a duplicate.

closed 2022-10-17 by rtyr

vdubedout, 2022-10-17:
Yes it looks like the same issue. Thanks 👍