prusa3d/PrusaSlicer #9054: First layer lines spacing

lolowl-coder, 2022-10-16:
**Is your feature request related to a problem? Please describe.**
I have a problem of first layer like this (not my picture, just taken for reference):
[image]
First lines are OK, but later plastic squeezes to the side leading to irregularities in lines and later plastic refuses to stick to the bed at all.

**Describe the solution you'd like**
Add a parameter like "first layer spacing" that controls distance between lines of first layer.
Perhaps in "Printer settings->Advanced->Other

**Describe how it would work**
Relative (to nozzle size) or absolute value.
Example:
relative value:
100.0% means distance between lines centers equals to nozzle size.
absolute value:
0.4 means distance between lines centers equals to 0.4 mm.

**Describe alternatives you've considered**
No alternative

**Additional context**
I've figured out this is because of line spacing that is too small.
But unfortunately in PrusaSlicer there is no possibility to affect this parameter.
I've compared first layer generated in Cura and PrusaSlicer and noticed, that Cura vs PrusaSlicer generates lines with higher spacing.
[image]
With the same model sliced in Cura I've got a perfect first layer and my problem went away.

Guys, your slicer is great and I'm sure, with this feature it will be my favorite.

BigBen2003, 2022-10-18:
English: English: In Prusa Slicer there is a setting for the height of the first layer:
Deutsch: In Prusa Slicer gibt es eine Einstellung für die Höhe der ersten Schicht:
[image]

lolowl-coder, 2022-10-18:
**BigBen2003**, I've tried different values for first layer height without success. This is not related to the issue mentioned.

rtyr, 2022-10-19:
Your issue is most likely caused by incorrectly calibrated first layer on the printer side (nozzle too low, uneven surface, adhesion issues).

If you want extrusions to overlap less on the first layer, you should just move the nozzle a little bit higher (it is ender, so likely moving the Z endstop a little bit higher).

If it is not possible or you really want gaps between extrusions for some reason, you can set lower flow for the 1st layer like this.

[image]

Here you can find some info about flow math used in PrusaSlicer https://manual.slic3r.org/advanced/flow-math.

 I am closing this issue.

closed 2022-10-19 by rtyr

ray1015, 2023-12-15:
I've been suffering from this problem for a long time too, it a good feature, when will it be added

19504643, 2023-12-15:
I've been suffering from this problem for a long time too, it a good feature, when will it be added