prusa3d/PrusaSlicer #9055: Use asyncronious layers for different object parts.

vovodroid, 2022-10-17:
Currently support layers could be (and actually forced, at least for single extruder) printed asynchronously to the object layers.
Why not to do the same for object parts to save time?

[image]

We can print cylinder with thicker layers than sphere, can we?

bubnikv, 2022-10-17:
Apart from implementation difficulty, there are two principial problems:
1) Part cooling may be insufficient. This is IMHO a show stopper for many objects.
2) Collisions with the extruder assembly / gantry would need to be checked for.

vovodroid, 2022-10-17:
> 1. Part cooling may be insufficient.

Why? Is currently fan seed changed with layer height?

> 2\. Collisions with the extruder assembly / gantry

If you only rise, there is no such problem. And if you descent 0.4-0.2 mm - only nose tip/side angle should be taken into account.

closed 2026-09-29 by prusaslicer-bot[bot]