prusa3d/PrusaSlicer #9056: Native Klipper Support similar to SuperSlicer

disan87, 2022-10-18:
Hi Dev Team,

I would like to Klipper firmware support (with moonrake) similar to SuperSlicer. I would like to connect PrusaSlicer to my Voron 2.4 and send gcode over netzwork to my printer. 

Best regards,
Disan

rtyr, 2022-10-18:
We are planning to add separate "Klipper" flavor, but you can use "Marlin (legacy)" flavor for Klipper (as instructed on Klipper website). You can also send g-code files directly and use for example Mainsail UI.

disan87, 2022-10-18:
Will it be possibile to send g-code from slicer directly?

rtyr, 2022-10-18:
Yes, you can leave OctoPrint there and just fill in IP adress.

I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3452.

closed 2022-10-18 by rtyr