prusa3d/PrusaSlicer #9059: Gaps

ljungis, 2022-10-18:
### Description of the bug

Printing this part in PETG using Arachne with default settings doesn't close the gaps marked.
Which setting will I have to edit to close them?
Using the classic perimeter generator closes them.

### Project file & How to reproduce

[image]
[Washer.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/9812820/Washer.stl.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+linux-x64-GTK3-202209060725.AppImage

### Operating system

Linux Mint 20.3

### Printer model

Wanhao Duplicator i3  MK2

BigBen2003, 2022-10-18:
English: When the example file is sliced in Prusa Slicer 2.5.0 with the filament profile "Prusament PETG" and the original settings "0.3mm DRAFT", no gaps are generated:
Deutsch: Wenn die Beispiel-Datei im Prusa Slicer 2.5.0 mit dem Profil für das Filament "Prusament PETG" und den Original-Einstellungen "0.3mm DRAFT" gesliced werden, werden keine Lücken generiert:
[image]

ljungis, 2022-10-18:
BigBen2003, thanks for your reply. I see that the profile you are using results in 2 perimeters.
Mine givs 3 perimeters. If you add a perimeter what will that do.
When I set 2 perimeters my preview looks like yours.

BigBen2003, 2022-10-19:
Deutsch: Bei einem erneuten Test ist mir folgendes aufgefallen:
English: When I tested again, I noticed the following:
[image]

Deutsch: Es handelt sich hierbei um die erste Schicht. Diese wird ohne Lücken gedruckt. Alle weiteren Schichten werden mit der Lücke gedruckt. Alle Lücken befinden sich immer an der gleichen Stelle. Wenn das gedruckte Modell stark beansprucht wird, kann es sogar eine Bruchstelle darstellen, da an den Stellen immer eine Lücke vorhanden ist.
Mir ist weiter aufgefallen, dass die Lücken auch in der letzten Deckschicht vorhanden ist. Hier hätte ich eher angenommen, dass die Deckschicht wie die erste Schicht ohne Lücken gedruckt wäre.

English: This is the first layer. This is printed without gaps. All further layers are printed with the gap. All gaps are always in the same place. If the printed model is subjected to heavy use, it may even present a break point, as there is always a gap at the points.
I also noticed that the gaps are also present in the last top layer. Here I would have assumed that the top layer would be printed like the first layer without any gaps.

Deutsch: Es müsste eigentlich eine Einstellung im Slicer geben, mit der festgelegt wird, ob die letzte Deckschicht, wie die erste Schicht gedruckt wird. Vielleicht gibt es eine solche mir unbekannte Einstellung bereits. Selbst wenn es solche Einstellung gibt, ist immer noch nicht die Tatsache aus der Welt geschaffen, warum an den beiden Stellen eine Lücke bleibt.

English: There should actually be a setting in the slicer that determines whether the last top layer is printed like the first layer. Perhaps there is already such a setting unknown to me. Even if there is such an attitude, the fact of why there is a gap in the two places is still not settled.

neophyl, 2022-10-19:
Usually the extrusion width for first layer is configured to be slightly thinner than others, for example 0.42 instead of 0.45.  This will account for why there are less gaps on the first layer as the perimeter extrusions are thinner and so they can be placed closer together and so the space in the pointy areas is less.  
If you want less gaps then you will need to set other extrusion widths to a smaller value too.

To the original poster, could you please include the requested project file and not a stl file.  A 3mf project will contain all your settings and allow anyone using it to slice it exactly as you are doing.  It removes the guess work.

ljungis, 2022-10-19:
[Fågelmat_2-Bricka.3mf_FILES.zip](https://github.com/prusa3d/PrusaSlicer/files/9821649/Fagelmat_2-Bricka.3mf_FILES.zip)
Thank's to both of you. 
In my case the gaps are present in every layer and I can see thru the printed part.

neophyl, 2022-10-19:
I'm sorry but the zip file does not contain any valid project.  Extracting it as a proper zip doesn't have a 3mf inside it and renaming it as a 3mf doesnt open in PS.

Just use file>Save as in PS to generate a PS 3mf project file and then zip it up and attach it.

ljungis, 2022-10-19:
Hi and sorry but putting this .3mf file in a zip archive doesn't work for
me.
So I send is this way.

Regards
ljungis

neophyl, 2022-10-19:
I don’t think you can send attachments like that through GitHub. It’s why you have to zip and attach or use something like Dropbox with a shared file. Zip and attached to an actual post on the website is always preferred though.

ljungis, 2022-10-19:
I didn't manage to zip it, so here is a link to my Dropbox where it can be downloaded.

https://www.dropbox.com/s/k32sdykf1rinupe/F%C3%A5gelmat_2-Bricka.3mf?dl=0

ljungis, 2022-10-19:
Or will this work??

[Fågelmat_2-Bricka.3mf_FILES.zip](https://github.com/prusa3d/PrusaSlicer/files/9824589/Fagelmat_2-Bricka.3mf_FILES.zip)

ljungis, 2022-10-20:
Or this
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/9826729/PrusaSlicer.zip)

closed 2026-09-29 by prusaslicer-bot[bot]