prusa3d/PrusaSlicer #9063: Wipe into this object switches only to extruder 1

boristomas, 2022-10-19:
### Description of the bug

  When switching tools on slice preview if object has Wipe into this object set, app switches to Extruder 1 (only) and ignores user set tool changes

### Project file & How to reproduce

When switching tools, on slice preview if object has Wipe into this object set, app switches to Extruder 1 (only) and ignores user set tool changes

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.3.0

### Operating system

Windows 10 x64

### Printer model

Prusa MK2 + MMU2

lukasmatena, 2022-10-19:
Provide a 3MF. Thanks.

boristomas, 2022-10-20:
> Provide a 3MF. Thanks.
Sure, here you go:
[outside unit wall plate.zip](https://github.com/prusa3d/PrusaSlicer/files/9827697/outside.unit.wall.plate.zip)

PxT, 2024-07-13:
Just ran into this today, PrusaSlicer 2.8.0, MK3S
Selecting "Wipe into this object" makes the whole object use tool 1.
[Sample.3mf.zip](https://github.com/user-attachments/files/16203321/Sample.3mf.zip)

closed 2026-09-29 by prusaslicer-bot[bot]