prusa3d/PrusaSlicer #9064: cant zoom out long enough

boelle, 2022-10-19:
### Description of the bug

what is the problem: cant zoom out enough
what did you expect: zoom out so that the whole printer/bed model can be seen

### Project file & How to reproduce

my printer model is a 71 mb stl so could not upload, not even as zip

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

win 12

### Printer model

franken build with mix's of mendel90 and prusa

kubispe1, 2022-10-20:
Really interesting idea. What is the usecase of importing whole model with encloser to Bed shape settings?

boelle, 2022-10-20:
It was just the fastest thing to do

Can of course export less

closed 2026-09-29 by prusaslicer-bot[bot]