prusa3d/PrusaSlicer #9066: Make STL repair work on Linux

kozross, 2022-10-19:
**Is your feature request related to a problem? Please describe.**

Currently, Linux installations of PrusaSlicer do not have access to the Netfabb-driven STL repair feature available on Windows. Its use is illustrated below for clarity:

[image]
[image]

For clarity, these are illustrations provided by a Prusa Research support staffer during diagnosis of a strange slicing result, which turned out to be a model error. I could not make use of this feature as a Linux user at all. Worse, I even attempted to run PrusaSlicer under Wine, and it _still_ didn't work: clicking the orange warning triangle silently does nothing.

Setting aside that silent failure is definitely not desirable UI behaviour here, this effectively relegates Linux users (and I assume, Mac users) to second-class status vis a vis feature support. While I realize that PrusaSlicer is, above all, a _slicer_, the fact is that this feature already exists and is accessible to Windows users, and Windows users only. This runs directly counter to the claim by Prusa Research that PrusaSlicer supports Linux; to my reading, this implies feature parity.

**Describe the solution you'd like**

The following UI change, which I consider necessary in any case:

- If this feature is unavailable or fails to work, a clear indication, by means of a dialogue box or similar, that the repair failed, and stating why (wrong platform, say).

Additionally, one of the following:

- Support for either this exact feature, or something similar, on non-Windows platforms;
- Specific instructions for how you can run PrusaSlicer under Wine to get the same capabilities, if platform ties are the problem;
- A clear statement that support for Windows, and _only_ Windows, is first-class, and other platforms are second-class, with what features are missing, somewhere quite visible.

**Describe how it would work**

The UI change would help considerably, especially for new users. Depending on what solution is used, the outcomes would be different:

- Support everywhere would be the ideal, and would bring the experience of users on non-Windows platforms in-line with those on Windows.
- Being able to emulate PrusaSlicer's capabilities in this regard consistently and in a documented manner under Wine would be the next best thing, as this would at least allow this feature to be accessible, if in a somewhat more involved way.
- Clearly stating that Linux is second-class (along with Mac, I assume) would make it clear that we shouldn't expect this to work. It would also help Prusa Research support staff.

**Describe alternatives you've considered**

Obviously, manually fixing this using Blender or similar software is possible, but compared to what a Windows user of PrusaSlicer would have to do, this is _incredibly_ onerous.

lukasmatena, 2022-10-20:
Thanks for the suggestion. The reason why the NetFabb fixing is not available in Linux/macOS builds is that the whole repair service is in fact provided by Microsoft itself, and it is simply only available in Windows 10 and above.

We are well aware of the fact that the feature is missing on the other platforms. In fact, we already tried integrating a different mesh fixing algorithm to substitute NetFabb on Linux and macOS, but the results were not satisfactory. We may get back to it in future. Sadly the problem of fixing a broken mesh is quite difficult and we can be grateful that the service is provided on Windows.

Regarding your other suggestions:
- UI: we may discuss this internally. Adding something to inform the user about why the feature is not available is a possibility.
- Wine: I don't think that it can possibly work on Wine.

kozross, 2022-10-20:
@lukasmatena - Thanks for the feedback. While I am a bit sad that it makes Linux (and Mac) users of Prusa's official slicer second-class, I also understand there's not a lot you can do about it.

Out of interest, is the service provided by Microsoft an API callout of some kind? Netfabb for Linux _does_ exist (in a much older version), and can provide similar fixes. However, I assume calling out to it is either impractical or impossible.

bubnikv, 2022-10-20:
> Out of interest, is the service provided by Microsoft an API callout of
some kind? Netfabb for Linux *does* exist (in a much older version), and
can provide similar fixes. However, I assume calling out to it is either
impractical or impossible.

Microsoft provides a fixing service through Windows 3D Printing API. We
actually don't know what algorithm Microsoft uses and where they got them.
We assume they bought the algorithms from Netfabb (which is now part of
AutoDesk), we may however be wrong and also the situation may have changed
over the years. For example, it seems that initially Microsoft ran the 3D
fixing service in the cloud, so your Windows box sent your 3D data to
Microsoft for processing. Nowadays it seems the same Windows API processes
the data locally.

BillyCroan, 2023-01-27:
I'd love to see this working.  It's rather disheartening to have to throw away my values for software library just to fix a mesh.

Some have mentioned a netfan online service.  Link?

Is there no Linux command line tool to fix meshes?

neophyl, 2023-01-27:
There used to be a cloud based netfabb fixing service but it was closed.  So a link is no longer available.  Autodesk do have a habit of folding previously free tools/functionality into paid for versions of software.

kozross, 2023-01-27:
@neophyl Also of hating Linux to the point of removing support from existing products that had it.

Dapid, 2024-06-23:
I don't have extensive experience, but I found admesh to be able to fix anything I threw at it, and it is blazing fast.

(The author claimed to be able to fix a mesh with 320 000 facets in 13 minutes, my 266 588 facets on my modern laptop takes 0.15 s, but I do have a bit more than 16 MB of memory).

kozross, 2024-06-24:
@Dapid - link?

Dapid, 2024-06-24:
Oops, I forgot to include it: http://github.com/admesh/admesh/

Dire-Pants, 2025-01-31:
Meshlab (FOSS for Linux) also has repair functions under the "Filters" section.

merll002, 2025-08-25:
Thanks @Dapid for the admesh suggestion. It works really well!
For anyone wondering how to use it, just:
1. Download the latest release
2. Extract
3. Run `./configure`, then `make`
4. Install with `sudo make install`
5. Repair with `admesh <input stl> -b <repaired stl>`

kozross, 2025-10-01:
@jokeyrhyme - this is unsurprising. This functionality was forked 100% unchanged by Bambu, and then forked unchanged _again_ by Orca.

closed 2026-09-29 by prusaslicer-bot[bot]