prusa3d/PrusaSlicer #9067: Proper multiple extruder selection -- separate brim extruder from perimeter/wall extruder

JoshLRR, 2022-10-20:
Separate brim and perimeter extruder in multiple extruder settings. It is so dumb that I cannot print soluble brims without also being forced to print my part with soluble walls. 

I just want to be able to choose a separate extruder for a brim. I don't want to print my walls out of the same material as my brim. Just WHY was it implemented like this?

DragonFinlay, 2022-10-28:
Usually, if something is requested a lot but not implemented over a longer period, there's a good reason for it.

Examples of such reasons would be:
- Implementing this would interfere with another major part of how the software currently works
- It's either too time-consuming or the priority is too low, or a combination of both

Currently, PrusaSlicer calculates skirt and brim for all objects at once, mainly because they need to be connected and recalculated if objects are close to each other.
Changing this to allow using a different extruder for them would mean that we have to significantly change the algorithm, potentially losing other functionality that's more important for us. It's always a trade-off.

So please stay patient, I'm sure we will be able to implement this one day, but for now we need to keep it on our to-do list.
Feel free to slice this part with SuperSlicer, iirc it does have this functionality, or at least something similar to it.

Snuff1eupagus, 2024-05-02:
@JoshLRR At the time you posted this, it was already possible to do this for a little over a year.

This may help you.

https://www.youtube.com/watch?v=dhsbMmosbLU

rob-miller, 2024-10-26:
> @JoshLRR At the time you posted this, it was already possible to do this for a little over a year.
> 
> This may help you.
> 
> https://www.youtube.com/watch?v=dhsbMmosbLU

Says video unavailable?

Same as #10491 ?

hapiel, 2025-01-19:
> > https://www.youtube.com/watch?v=dhsbMmosbLU
> 
> Says video unavailable?
> 

Link no longer works, but here is a new one: https://www.youtube.com/watch?v=IEKUpZ-gjMs

dsanders2, 2025-04-07:
> > > https://www.youtube.com/watch?v=dhsbMmosbLU
> > 
> > 
> > Says video unavailable?
> 
> Link no longer works, but here is a new one: https://www.youtube.com/watch?v=IEKUpZ-gjMs

The video so no longer available again.

closed 2026-09-29 by prusaslicer-bot[bot]