prusa3d/PrusaSlicer #9068: Stitch the layers using injection molded rivets

Harvie, 2022-10-21:
Hello guys,
i have this idea to increase layer adhesion and prevent delamination.
Basicaly you create cavity which is used as injection mold to create "rivet" bonding several layers together.
This will probably need to increase temperature so the filament flows more smoothly into such mold.
What do you think?

[image]

Harvie, 2022-10-21:
Viable injection depth is obviously limited, but we can use interleaved pattern like this:

[image]

murk-sy, 2022-10-23:
Devs are more likely to get hooked on an idea if they can print a practical example to test, which means you'll need some test gcode/3mf. Or at least make a test manually, like printing a model with a tall hole and see how far (or if) the extruder can push the material when you operate it above the hole manually.

I have an idea how to make a test *relatively* easily (model with holes and then just add extrusion on alternating points manually every layer change), but I'll see if I have time.

Personally, I think this is unlikely to improve layer adhesion compared to just adding another perimeter, which is also likely to be faster. But definitely interesting enough to test at some point.

Honestly, I'm more tempted to test out injection molding really small parts using just a regular 3d printer hotend. Probably not gonna work, but I'm really curious.

DragonFinlay, 2022-10-28:
I find it interesting as well, but as murk-sy mentioned, it would be important to have some data on this, showing that it actually has a noticeable impact on layer adhesion, plus a test file for us to print.

Harvie, 2022-10-28:
Haha. There is just one tiny problem. I don't have 3d printer :-D and i will not have one soon, because i am kinda on the move right now. Is there anybody willing to prepare such test? I had crazy idea of contacting one of the youtubers who are making living by doing exactly these tests, but it's not easy to cold-approach someone with so many followers.

jflyer81, 2022-11-02:
Maybe reach out to Stephan at CNC Kitchen? This seems like the kind of thing that he would be interested in, and I've also found him to be pretty open to talking with people about interesting project ideas.

https://www.cnckitchen.com/impressum

Harvie, 2022-11-02:
> Maybe reach out to Stephan at CNC Kitchen?

That is exactly what i did :-) But i didn't knew he had a contact form, so i've tried to PM him on the reddit. But now i've also filed the form, so we'll see.

murk-sy, 2022-11-02:
I did some napkin math assuming the best-case-scenario that will probably be discouraging. tl;dr at the end.
File is attached if you want to play around with it.

For this I made the following assumptions:
- Only tensile strength is calculated
- XY tensile strength is universal for injection molded parts
- Z / layer adhesion tensile strength is 50% of XY tensile strength
- "normal" outer perimeters have layer adhesion tensile strength
- "improved" innermost perimeters have XY tensile strength (i.e. "perfect")

[image]

Rather than trying to use injection molded rivets (which would have a very small area and require extra perimeters to be enclosed), I've made an assumption that we're basically just printing entire "improved" perimeters with the best possible tensile strength. A single perimeter would cover a much larger area than any reasonable amount of rivets.

However, the conclusion is: *You're better off just adding an extra perimeter.* Comparing a 3 perimeter print with a 2 perimeter print + an entire perimeter's worth of rivets, you're better off making just 4 perimeters from the get to. **Keep in mind that this comparison assumes that basically 30% of all printed area in a layer is rivets.**
And even assuming you have a perimeter's worth of rivets, **you still need extra printed geometry to make holes for those rivets**. That would add a lot of extra circles to print (meaning a bunch of slowdowns), not even counting the actual injection process itself.

Based on my experience making (mainly small) technical parts, I've often had print time reduced by adding extra perimeters and not changing any other settings. Using this process, however, would greatly slow down the print overall.

Spreadsheet: [2022-11-02 layer stitching prusaslicer issue 9068.ods](https://github.com/prusa3d/PrusaSlicer/files/9922548/2022-11-02.layer.stitching.prusaslicer.issue.9068.ods)

**TL;DR:** Add 2 more perimeters for greater improvement than rivets could probably ever provide, and for a much smaller time investment.

DragonFinlay, 2022-11-04:
Thank you so much for looking into this and providing some numbers, interesting to see.
I'll keep this open in case someone else is interested to try, perhaps make a post-processing script for simple tests.

mejames7760, 2023-10-21:
Has anyone looked into shaking extrusion lines in xy?  Without doing any detailed analysis I started putting little squares around the perimeter of pipes intersecting with planes because it seemed to reduce breakage at the intersection. The idea being to interrupt the line of fracture.  This is not a really good idea because the inner perimeter lines don't follow the extrusion and larger blocks get in the way of hoses attached to the pipes so the fractures happen after the blocks in z. (assuming that the pipe is xy rings stacked in z, which is the way most of my pipe designs end-up) But if all the perimeter lines are randomly shaking around the xy direction vector this could also break-up the stress fracture line some. It would seem to need firmware support to accomplish.

Harvie, 2023-10-22:
> Has anyone looked into shaking extrusion lines in xy? 

I think it would make more sense to shake the lines up and down in Z. it would improve contact patch between two layers.
eg.: when lines are rippled like this, they obviously have more contact area between two layers than they would have if they would be straight. might have negative visual effect on surface finish tho...

[image]

k1-801, 2023-10-22:
I can see a use for direct plastic injection feature, although not in a form of rivets.
It could be used to encase embedded nuts when those are placed in a vertical plane, flat side horizontal. With a hexagonal nut, this approach leaves two unfilled triangles on top of two inclined sides of the nut (only one if we make a tilted "channel" to drop the nut in), that can not be filled during the print in any way other than injecting a certain amount of plastic in there right after the nut is placed, with little to no head motion. It might even be a good idea to overheat the plastic in the area slightly just so it flows better when touching the cold freshly placed nut.

murk-sy, 2024-11-22:
Only just found out it's patented (or at least a similar thing is)

Video where I saw it: https://www.youtube.com/watch?v=TbYa7BGvtSw
https://patents.google.com/patent/EP4100235B1/en

So apparently it works better than I assumed, but I would guess it's mostly effective when already printing solid parts.

CNCKitchen, 2025-01-24:
> > Maybe reach out to Stephan at CNC Kitchen?
> 
> Contacting [@CNCKitchen](https://github.com/CNCKitchen) is exactly what i did :-) But i didn't knew he had a contact form, so i've tried to PM him on the reddit. But now i've also filed the form, so we'll see.

Took two years to get this notification in my inbox. As @murk-sy already said, AIM3D implemented this in their slicer and it seems to improve strength for basically solid printed parts. Unfortunately they also filed a patent but I'll hope to have a chat with them the next weeks about the situation.

CNCKitchen, 2026-02-02:
Yes, I did last year at Formnext. They were very open about all of this. Even though I still don't agree with their patent filing (yet I'm not a lawyer), I had the impression that they are not trolls and won't come after others using/playing around with the technology, and would rather appreciate it if people prove its feasibility.

Harvie, 2026-09-02:
still relevant