prusa3d/PrusaSlicer #9072: Better seam hiding on cylinder-shaped objects [Feature Request]

laptop-laptop, 2022-10-22:
**Is your feature request related to a problem? Please describe.**

When printing elongated rounded object like a cylinder or bottle, all available options cannot hide the seams.
Random and Nearest seem to produce similar results where clusters of seams are spread on the rounded surface
like fireworks on the sky. As for the other options, they just product an ugly line of seams.

**Describe the solution you'd like**

A solution that would allow better hiding of seams for this type of rounded objects.

rtyr, 2022-10-22:
What solution do you propose? Where do you want to place the seams?

laptop-laptop, 2022-10-22:
Can the Fuzzy skin feature somehow be modified to have ironing on the surface that could iron out the holes due to the seams?
Alternatively, is it possible to start and end each layer a bit inside the cylinder-type object so the opening is hidden inside the cylinder?

murk-sy, 2022-10-25:
I think if anyone found a reasonable way to hide the seam it would be implemented already. The issue is start/stop of the extrusion, which is why vase mode has no seams (not counting some artifacts with current vase mode code).

I proposed a spiral seam option some time back in #6689 - that would make it look more like a feature rather than a defect, but it would still not hide it. It would, however, make any seam bumps on things like wheels imperceptible.

DragonFinlay, 2022-10-28:
Yeah, seams on objects with no hard edges have always been a bit of a struggle. Under specific circumstances (smaller nozzle, matte filament with particles, etc.) I find random to be the best option since the seams will already be quite invisible.
But for bigger nozzles and glossy filaments it's a pain to try and hide them.

I think the only solution is to have a technique to better hide the extisting seams. What @laptop-laptop proposed is already somewhat being done with wiping before retraction, but it's not perfect either of course.

MichaelJLew, 2023-11-08:
I have a fix for seams on cylinders (and for any shaped object) that you might find interesting. I have not yet put together a GitHub post, but have a preliminary report on the PrusaSlicer forum: https://forum.prusa3d.com/forum/prusaslicer/better-seams/#post-683251

MichaelJLew, 2023-11-08:
See here: https://github.com/prusa3d/PrusaSlicer/issues/11621

SaschaKP, 2023-11-30:
as proposed by @MichaelJLew the solution is with a lower height start, lower layer height than what is set, maybe a factor of 2 (so 0.1 at start, 0.2 middle of the line, 0.1 at end, incremental), with an higher end, lower layer height, that would hide the seam, the longer the line, the harder it will be to see the seam, square or round, it doesn't really matter. This will of course cause a little longer printing time, but the appearance is really better than what we have now.

MichaelJLew, 2023-11-30:
@SaschaKP That's not quite how the scarf seam should work. Start at zero layer height and taper up to the nominal layer height. If you start with half layer height you will still get a seam. See [A method for less visible seams](https://github.com/prusa3d/PrusaSlicer/issues/11621) for the full explanation and en example file.

SachCZ, 2025-01-21:
Scarf seam are in PrusaSlicer 2.9.0. Closing.

closed 2025-01-21 by SachCZ