prusa3d/PrusaSlicer #9074: Arachne attempts to bridge before adding filament bridges connect to

DFEvans, 2022-10-22:
### Description of the bug

With a model where bridges are connected to additional, adjacent extrusions, the Arachne perimeter generator will sometimes attempt to extrude the bridges first, not connecting to any adjacent filament, or filament in the layer below.

I have the following layer in an object, where the dark blue bridging section connects to adjacent areas of infill:

[image]

If I scroll through the GCode preview for the layer, it shows that it will attempt to print these bridging sections in thin air, isolated from the layer underneath, and before printing the adjacent filament the bridges would connect to:

[image]

In contrast, the Classic perimeter generator will print the un-bridged sections first:

[image]

### Project file & How to reproduce

[BridgingInThinAir.zip](https://github.com/prusa3d/PrusaSlicer/files/9845225/BridgingInThinAir.zip)

Slice the model, scroll to layer 27 in the preview, and scroll the GCode preview to near the start of the layer.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10

### Printer model

Original Prusa Mini+

MrBlackmidi, 2022-11-19:
i feel like arachne is really weird in prusaslicer as of right now

kubispe1, 2023-04-20:
Retested with **2.6.0-alpha6 + 100** issue is still there.
[image]
[image]

fluksh, 2023-05-27:
Same problem in 2.6alpha6, it is not even recognizing the free floating extrusions as bridges

[image]

BloodyRain2k, 2023-07-26:
I've run into the same problem with the 2.6 release version:
[image]
[image]
I've already tried disabling various experimental settings that work fine everywhere else, but the only difference was the preview's line color changing from blue to orange.
Still tried to start the layer mid-air regardless : (
Here's the project with the settings I tried.
[Soap_Stand.zip](https://github.com/prusa3d/PrusaSlicer/files/12174795/Soap_Stand.zip)

tuct, 2023-09-03:
Any news on this, I still see this in 2.6.1 and hate falling back to classic besides the fact that usually I only realize the issue when printing not before.

MrBlackmidi, 2023-09-14:
Afaik, arachne is low priority rn

nicole-ashley, 2023-10-25:
Why would Arachne be a low priority, isn't it the default? This is a pretty glaring issue for a default setting.

Here's another example, slicing [this model](https://www.printables.com/model/161938-joycon-controller-with-trigger-buttons-easy-print) at 0.2mm layer height on a 0.4mm nozzle with Arachne (the only setting I can find that changes the behaviour): 

Layer 199:
[image]

It's not always the case, for example layer 330: 
[image]

tastyratz, 2024-03-13:
I did a lot of digging on this issue and narrowed it down in the linked orcaslicer bug and reference the same bug downstream on superslicer as well.
This has to do with Arachne wall order when more than 1 perimeter is selected.

The solution would be "print detected bridges last/after perimeters" as a default behavior or checkbox for when bridges are detected. This should drastically improve bridging performance.

nicole-ashley, 2024-03-13:
@tastyratz You have indeed put a lot of work into figuring this out which I very much appreciate! Hopefully it's enough to get a fix in one of the Slic3r derivatives that can easily be ported across all of them.

WhiterRice, 2024-04-21:
Adding some simple recreation steps,

Model: https://www.printables.com/model/116915-bridge-test-piece
Prusa Slicer 2.7.4
Mk3s default 0.4 Profile 0.2mm quality
Project FIle: [Issue_9047.zip](https://github.com/prusa3d/PrusaSlicer/files/15053098/Issue_9047.zip)

Arachne Perimeter:
[image]
Classic Perimeter:
[image]

sslupsky, 2026-09-02:
This is a significant issue that results in failed prints.

tastyratz, 2026-09-03:
This issue appears to still be happening as mentioned in a related issue, #11430 

Please do not close this priority bug.