prusa3d/PrusaSlicer #9078: Artifacts near sharp corners

vovodroid, 2022-10-23:
### Description of the bug

Strange artifacts appear near sharp corners

[image]

[Stopper.zip](https://github.com/prusa3d/PrusaSlicer/files/9846542/Stopper.zip)

### Project file & How to reproduce

Slice attached project.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Win10x64

### Printer model

Biqu B1 SE

vovodroid, 2022-10-23:
Actually it's not just sharp corner, but arc tangential to line, so object width is going to zero. After applying 0.1 mm fillet artifact disappeared.

DragonFinlay, 2022-11-01:
This is the same issue as #8730, the features are just too thin to be considered by the perimeter generator.
Feel free to see if the proposed solution in the comments helps (decreasing "Minimum perimeter width") and feel free to comment there.

Closing as duplicate.

closed 2022-11-01 by DragonFinlay