prusa3d/PrusaSlicer #9079: Prusa linux build instructions fail finding several libraries

src-r-r, 2022-10-23:
### Description of the bug

Using the [Linux Build Instructions](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md) During the Prusa make stage, Pursa is unable to find many of the libraries that it built for dependencies.

The following error is produced (relevant portions):

```
....
                                      
-- EXPAT found in /opt/PrusaSlicer/deps/build/destdir/usr/local/lib/cmake/expat-2.4.3                                                 
-- GLEW found in /opt/PrusaSlicer/deps/build/destdir/usr/local/lib/cmake/glew                                                         
CMake Error at cmake/modules/FindNLopt.cmake:124 (message):                                                                           
  Unable to find requested NLopt installation: Cannot find NLopt library                                                              
  'nlopt'.  NLopt not found in system directories (and environment variable                                                           
  NLOPT is not set).                                                                                                                  
Call Stack (most recent call first):                                                                                                  
  CMakeLists.txt:548 (find_package)                                                                                                   
                                                                                                                                      
                                                                                                                                      
-- Configuring incomplete, errors occurred!                                                                                           
See also "/opt/PrusaSlicer/build/CMakeFiles/CMakeOutput.log".                                                                         
-- SLIC3R_ASAN: OFF                                                                                                                   
-- SLIC3R_BUILD_SANDBOXES: OFF                                                                                                        

....
                                                   
CMake Error at cmake/modules/FindNLopt.cmake:124 (message):                                                                           
  Unable to find requested NLopt installation: Cannot find NLopt library                                                              
  'nlopt'.  NLopt not found in system directories (and environment variable                                                           
  NLOPT is not set).                                                                                                                  
Call Stack (most recent call first):                                                                                                  
  CMakeLists.txt:548 (find_package)
```

After doing some digging this is caused by `PrusaSlicer/cmake/modules/FindNLopt.cmake:31`

```cmake
	set(_NLopt_LIB_NAMES "nlopt")
```

If I inspect `${PRUSA_CMAKE_PREFIX}/lib`, I find a file `libnloptd.a` (the same goes for `libnlopt_cxx`). So I need to make the following change to `PrusaSlicer/cmake/modules/FindNLopt.cmake` (lines 31 and 75 respectively for it to continue:

```cmake
# line 31
	set(_NLopt_LIB_NAMES "nlopt" "nloptd")
# line 75
	set(_NLopt_LIB_NAMES "nlopt_cxx" "nloptd_cxxd")
```

However, there are still issues I'm grappling with. I'm not getting the error:

```
CMake Error at CMakeLists.txt:560 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.
```

If I add some debugging to `FindOpenVDB.cmake` however it seems to do a pretty decent job finding everything...it just marks OpenVDB as not found:

```
-- OpenVDB_ABI=8
-- OpenVDB_DEBUG_SUFFIX=d
-- OpenVDB_DEFINITIONS=-DOPENVDB_ABI_VERSION_NUMBER=8
-- OpenVDB_FIND_COMPONENTS=openvdb
-- OpenVDB_FIND_MODULE=1
-- OpenVDB_FIND_REQUIRED_openvdb=1
-- OpenVDB_FIND_VERSION=5.0
-- OpenVDB_FIND_VERSION_COMPLETE=5.0
-- OpenVDB_FIND_VERSION_COUNT=2
-- OpenVDB_FIND_VERSION_EXACT=0
-- OpenVDB_FIND_VERSION_MAJOR=5
-- OpenVDB_FIND_VERSION_MINOR=0
-- OpenVDB_FIND_VERSION_PATCH=0
-- OpenVDB_FIND_VERSION_TWEAK=0
-- OpenVDB_FOUND=FALSE
-- OpenVDB_INCLUDE_DIR=/opt/PrusaSlicer/deps/build/destdir/usr/local/include
-- OpenVDB_INCLUDE_DIRS=/opt/PrusaSlicer/deps/build/destdir/usr/local/include
-- OpenVDB_LIBRARIES=OpenVDB_openvdb_LIBRARY_RELEASE-NOTFOUND
-- OpenVDB_LIBRARY=
-- OpenVDB_LIBRARY_DIRS=
-- OpenVDB_LIB_COMPONENTS=OpenVDB_openvdb_LIBRARY_RELEASE-NOTFOUND
-- OpenVDB_MAJOR_VERSION=8
-- OpenVDB_MINOR_VERSION=2
-- OpenVDB_PATCH_VERSION=0
-- OpenVDB_USES_BLOSC=TRUE
-- OpenVDB_VERSION=8.2.0
-- OpenVDB_openvdb_FOUND=FALSE
-- OpenVDB_openvdb_LIBRARY=OpenVDB_openvdb_LIBRARY_RELEASE-NOTFOUND
-- OpenVDB_openvdb_LIBRARY_DEBUG=/opt/PrusaSlicer/deps/build/destdir/usr/local/lib/libopenvdbd.a
-- OpenVDB_openvdb_LIBRARY_RELEASE=OpenVDB_openvdb_LIBRARY_RELEASE-NOTFOUND
```

I'm currently digging into why. It seems like openvdb complains if it finds the debug version of the library. I'll see if I can copy the variable to OpenVDB_openvdb_LIBRARY to see if that helps.

Is there a snapshot of Prusa I should be use for building that doesn't have these issues? As soon as I get these fixed I'll be submitting a PR.

### Project file & How to reproduce

I'm building on docker and including the Dockerfile I'm using to build.

```dockerfile
FROM ubuntu:kinetic

RUN apt update && apt install -y \
	git \
	build-essential \
	autoconf \
	cmake \
	libglu1-mesa-dev \
	libgtk-3-dev \
	libdbus-1-dev

############################################
#
# Common environment variables for setting up exertnal deps.
#

ENV OPT_DIR /opt

# Ensure opt is empty!
# RUN rm -vfr ${OPT_DIR}/*

##########################################
#
# Environment variables for Prusa
#
ENV PRUSA_GIT=https://github.com/prusa3d/PrusaSlicer.git
ENV PRUSA_GIT_HASH=c3bdf2fc28bec3c1af816447c3a2b056009cfefe
# 2.5.0 release
# ENV PRUSA_GIT_HASH=ec2f533e3a355a2452bf6bd551da84020c69d8f8
ENV PRUSA_SRC ${OPT_DIR}/PrusaSlicer
ENV PRUSA_BUILD ${PRUSA_SRC}/build
ENV PRUSA_DEPS ${PRUSA_SRC}/deps
ENV PRUSA_DEPS_BUILD ${PRUSA_DEPS}/build
ENV CMAKE_PREFIX ${PRUSA_DEPS_BUILD}/usr/local
ENV PRUSA_CMAKE_PREFIX=${PRUSA_DEPS_BUILD}/destdir/usr/local/

# Set default git branch to main
RUN git config --global init.defaultBranch main

#########################
#
# Fetch prusa
#

WORKDIR ${PRUSA_SRC}
RUN git init ${PRUSA_SRC} || :
RUN git remote add origin ${PRUSA_GIT} || :
RUN git fetch origin ${PRUSA_GIT_HASH} --depth=1 || :
RUN git reset --hard FETCH_HEAD || :

########################
#
# Pursa dependencies
#

WORKDIR ${PRUSA_DEPS_BUILD}
RUN cmake ${PRUSA_DEPS} \
	-DCMAKE_BUILD_TYPE=Debug \
	-DDEP_WX_GTK3=ON
RUN make

########################
#
# Build prusa slicer.

WORKDIR ${PRUSA_BUILD}
RUN cmake .. \
	-DSLIC3R_STATIC=1 \
	-DSLIC3R_GTK=3 \
	-DSLIC3R_PCH=OFF \
	-DSLIC3R_WX_STABLE=ON \
	-DSLIC3R_GUI=no \
	-DCMAKE_PREFIX_PATH=${PRUSA_CMAKE_PREFIX}

RUN make -j 4 && make install
```

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

ec2f533e3a355a2452bf6bd551da84020c69d8f8

### Operating system

Ubuntu 20.04 host, Ubuntu 22.10 container

### Printer model

I don't have a printer.

tamasmeszaros, 2022-11-01:
Hi @src-r-r  I suggest not to build the dependencies in debug mode. If you want to debug inside library code, you might get away with RelWithDebInfo.

guysoft, 2023-12-17:
Hey, same in Ubuntu 23.04 with version 2.7.1. Is there a workaround for this?

mmiscool, 2024-03-14:
I would also like to find a working dockerfile that is setup to build prusaslicer from source targeting ubuntu x86-64. 

Dose any one have a working dockerfile that works to build?

amatulic, 2025-03-31:
I'm experiencing the same issue trying to compile 2.9.1 under Windows using Visual Studio 2022. The build script aborts with "OpenVDB could not be found with the bundled find module. You can try to specify the find module location of your OpenVDB installation with the OPENVDB_FIND_MODULE_PATH cache variable."

I have no idea what that means, or why I could compile 2.9.0 without issues. This 2.9.1 is a fresh repo clone.

closed 2026-09-29 by prusaslicer-bot[bot]