prusa3d/PrusaSlicer #9081: Indicate areas of no travel

yaderhey123, 2022-10-23:
I have several parts that I make with flexible filament that have some disconnected pieces at various layers of the print. I understand that stringing is almost inevitable with such a print, yet the location of these strings directly effects the performance of the part. If there is any stringing in the pink circles, the mechanism doesn't work as well. Because they are disconnected, using the "Avoid crossing perimeters" option won't have any effect.

[image]

Potential solution:
It would be very useful to be able to indicate areas where you would like the printer to avoid travel. Even if it causes significant detours. For example, you can indicate that the printer should never travel outside of the part within the blue rectangles in the image below.

[image]

How it might work: 
I think that this could be implemented by placing representative volumes that overlap with the model as "Travel Blockers." Or use a similar tool like the support blocker painter tool to indicate specific perimeters that should never be crossed.

Alternatives considered:
As mentioned, the avoid crossing perimeters does not apply here. Even in layers where the separate sections are connected by a thin perimeter, the nozzle will still cross in inconvenient locations.

DragonFinlay, 2022-11-01:
I really like this idea! Especially the suggestion about using modifiers to block travel sounds like something that could be done.
I'll see what the team says about this.

kubispe1, 2023-01-20:
Hi, We follow up on some hints of your FR and produced automize avoiding the crossing of curled overhangs. This feature will be part of release 2.6.0.
At the moment I am closing the issue.

closed 2023-01-20 by kubispe1

jaggzh, 2023-08-17:
What was implemented that addresses this?