prusa3d/PrusaSlicer #9083: Set View Centre (to allow rotation of view about specific point on object/screen

Jaxx005, 2022-10-23:
Hi,

**Issue**:    Rotating the View around a object feature is not easy, especially when the object is long/tall and/or the view is zoomed in.

Currently, the only option (that I have found) is to Zoom out, rotate, pan, and then Zoom back in (to view the feature in question from a different direction).  That is not always an easy procress.

Suggested solution:    **add a "Set View Centre" command**.   It would be useful to be able to spin the view about a specific, user-selectable feature of an object (e.g.  rotate view about the end of a clip)

Perhaps double-left-click on an object's surface, to select a 3D point, about which the View then rotates.  

This could be done automatically perhaps: automatic View Centering when the user zooms in to look at detail (that is away from the centre of the bed or object).   

Could also have the option to rotate about the centre of the screen (at bed level height?) - automatically, and/or by user choice.

Regards
L

DragonFinlay, 2022-11-01:
Hi, the most common way in other 3D software is to use the mouse as the center point for orbiting the camera. So the place the mouse points at when you click to orbit is taken as the center point.
I think that would also be more intuitive than manually setting the point. What do you say?

I don't think we have an issue about this yet, only about zooming to the cursor position. If you agree with me, maybe you could rephrase your issue to what I described above and we'll keep it as a feature request.

Jaxx005, 2022-11-03:
Hi, 
Thanks for your response...

I agree with click-and-drag principle to both centre and rotate, certainly for 'approximate' viewing. But for rotating about a specific point in 3D space, I am not sure how that centre could be selected by a single mouse click.  PS kind of suffers from this currently in a way...

Also (afaik), the Left mouse button is _currently_ used for View Rotation, but also for Object Selection & Object Move (so you would not be able to rotate about an object because left-click and drag only ever moves that object).  

One possible option [another feature request?]  would be to have a separate mouse-operation for View Rotation (e.g. scroll-wheel/middle-button, click and drag to rotate the view).  That would leave Left-clicking to only ever be for object selection, editing etc., which I think would be an improvement on its own :)

I would say some 3D CAD apps work in a similar way to this, but also have a more accurate 'Object-feature-as-centre' option too.

Note:  I have just found 'Z' (Zoom to Object) command:  it zooms to the active object - but also changes the centre of rotation to a point near the objects centre (sort of?).  But it seems to ignore height of the object, so you still have issues rotating about higher features/points (as well as features at ends of objects).   I don't see 'Z' in the app menus - I only found it in the online documentation [another feature request possibly!?] - see here:  

https://help.prusa3d.com/article/view_1761

I hope that makes sense!  I am totally happy to alter/simplify my original post as you see fit.  Shall I create a new comment for that or edit the original?

DragonFinlay, 2022-11-08:
I think this feature request is fine as is. We discussed both ideas here, our team will take a look and implement it when possible.

Jaxx005, 2022-11-15:
Great! Thanks.

Lukrative525, 2023-09-23:
I was just messing around with a long, thin part and thought of this as well. I've worked extensively with Solidworks, CATIA, and MasterCAM.

In both Solidworks and MasterCAM, the rotation point is reset each time the orbit command is activated based on where the mouse was when clicked. If the mouse is over the model, the center of rotation is set by projecting the point on the screen where mouse click occurred onto the nearest surface on the model. In CATIA, the rotation point is set by wheel clicking anywhere on the model, and is calculated the same way.

Either of these would be better in my opinion, just to make it so that the rotation point is more controllable.

pilot51, 2024-03-17:
Any update on this?

I find this to be the most annoying thing about PrusaSlicer because I'm often trying to look at details in an area of an object in the preview and it's very difficult to orbit around the area of interest in a predictable manner.

The 'Z' command doesn't really help other than snapping the orbit center to a known location. If I'm not looking at it perfectly top-down when I pan the view, the orbit center moves above or below the bed and becomes way more confusing if I wasn't intentionally trying to do that, which I'm usually not.

Without being able to select the orbit point with the mouse, it would be more intuitive for the orbit center to stay on whatever part of an object (including the bed) is at the center of the screen, never floating in nothingness. If the center of the screen is panned away from any object, I think it would make most sense to keep the orbit center at the last point it was on an object.

#6805 is closely related.