prusa3d/PrusaSlicer #9085: Arachne generator leaves gap in the print model

buckker, 2022-10-24:
### Description of the bug

I'm noticed this issue with my single wall parts for rc edf jets. The cad models have a 0.6mm outer shell and also 0.6mm thick ribs. The ribs have a small gap of 0.05mm to the outer shell with the goal that the slicer print the outer shell at once. 
With the standard perimeter generator the ribs are properly connected to the outer shell. With the new Arachne generator the ribs have a big gap to the outer shell and are not connected

Arachne:
[image]

Classic:
[image]

CAD solid:
[image]

Even better would be the way that Prusa Slicer print the outer shell at once without a gap of the ribs. But I'm not sure if this would be technically possible. 

Many thanks in advance

Michael

### Project file & How to reproduce

-

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Pruas Slicer 2.5 rc2

### Operating system

Windows 11

### Printer model

-

bubnikv, 2022-10-24:
The problem is not Arachne, but your model, where the inner walls are
disconnected from the outer walls.
You may try to increase the "closing radius" value, but in general it is
better to fix your model.

buckker, 2022-10-24:
Thanks for the response.

But why do I have no gap with the classic perimeter generator? And with the Arachne I do have gaps? I can't follow your argumentation. 

I would also like to design models without gaps. But the problem is that the outer shells won't be printed in a row. Especially with LW-PLA you get some ugly ooze artifacts. 

If I print the model as a solid the outer shell is printed in a row without any stops for printing the infill(the outer shell will be printed first, secondly the infill) . I like to have the same behavior for my single wall models. 

BTW, the closing radius parameter doesn't help.

PEZ18, 2022-12-06:
I have the same issue, and I dont understand the response bubnikv provided... This is happening with single walls, and multiple walls, so how can the cause be from inner and outer wall gaps in the stl design? I also tried adjusting the closing radius on my slicer, but it makes no difference. 

I also noticed random seams on external walls even though it is set to aligned. some of the external parameter defects are in the same location as seams, some are in the same locations as travel move points, and some I just no feature type indication. I included screenshots to show some of this as well.

Classic:
[image]
Arachne
[image]
Seam Locations Hidden (Arachne):
[image]
Seam Locations Shown (Arachne):
[image]

closed 2026-09-29 by prusaslicer-bot[bot]