prusa3d/PrusaSlicer #9086: Display area freezes while viewing sliced model

hiltonjanfield, 2022-10-24:
### Description of the bug

In "Preview" (Layers) view, the display area has been freezing on me frequently since the 2.5.0 final release. I did not experience this behaviour in the previous version (2.4.?), and I don't believe it happened in the 2.5.0 beta, but I may have just shrugged it off as a beta issue if it did.

Only the display area and controls within it (e.g. the View combo box) freeze; the entirety of the remainder of the GUI remains functional, which enables me to save my project, quit, and reload, the only solution I have found. The display will stop moving/responding, and if I switch tabs and go back, it will remain white. This also happens in "Settings in semi-modal window"  layout mode, excepting the turning white part.

I have not determined any particular action that is causing this to happen. it has happened while moving a view around, while moving the layer slider, while generating new gcode after changing settings, etc. It has happened with Background Processing both enabled and disabled.

I have not experienced this behaviour in 3D Editor view, only Preview.

It has happened with numerous project files and models of all sizes and complexity; nothing in particular seems to be triggering it. I initially thought I may be having issues with my laptop's graphics, but nothing else using them has displayed any similar behaviour, e.g. other slicers, 3D games, etc.

### Project file & How to reproduce

No project file included due to it happening on numerous projects. Nothing in particular seems to be triggering the freeze.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 11

### Printer model

Creality Ender 3 S1

DragonFinlay, 2022-11-01:
I can't say that I was noticing anything like that, not pre 2.5.0, not in 2.5.0. And of course it's difficult to comprehend an issue like this if you can't reproduce it. If you can capture this on video, it might help already, otherwise we'd need more info about your system.

My only assumption (other than the hardware or drivers) would be that it could be due to Windows 11 in general.

hiltonjanfield, 2022-11-03:
The only things I can add are:

- When it happens, I notice a brief system-wide pause (even my mouse cursor
freezes for a moment) so I am thinking it is a hardware or driver problem
and it has to reset, but the only thing affected afterward is Prusa's
display
- If I happen to have multiple instances of PrusaSlicer running, only the
current one will be affected.
- Once it happens, it seems to happen more and more frequently until I
reboot.

DragonFinlay, 2022-11-04:
> I notice a brief system-wide pause (even my mouse cursor freezes for a moment)
You mentioned that no other software has problems, but this does kinda sound like the graphics driver is crashing. Next time it happens, press Ctrl+Win+Shift+B to restart the driver and see if it helps.

> Once it happens, it seems to happen more and more frequently until I reboot.
That on the other hand sounds more like something might be overheating in your system. But again, the fact that nothing else freezes speaks kinda against that...

hiltonjanfield, 2022-11-19:
> You mentioned that no other software has problems, but this does kinda sound like the graphics driver is crashing. Next time it happens, press Ctrl+Win+Shift+B to restart the driver and see if it helps.

Unfortunately, this doesn't fix the issue. I still have to close and restart PrusaSlicer.

However, I _have_ noticed that it happens more frequently (or possibly only) when Fusion360 is also running.

DragonFinlay, 2022-11-29:
I see, thanks for the info. Unfortunately, it's difficult to reproduce and/or find out what exactly could be causing this without more info and screenshots/videos.

Please keep us updated here in case the issue persists with future versions.

hiltonjanfield, 2022-11-29:
Now that I'm paying attention to it, I will add that it definitely only happens when Fusion 360 is also running.

I have a similar issue with Cura (which I rarely use), it will routinely crash entirely but only with F360 running. At least only the preview in PrusaSlicer stops working so I can save my work!

bubnikv, 2022-11-30:
I am afraid we cannot do anything for you, if you have issues with both Cura and Fusion 360 on your system.

closed 2026-09-29 by prusaslicer-bot[bot]