prusa3d/PrusaSlicer #9087: Support on build plate only gives first layer supports with nothing above

brianbienvenu, 2022-10-24:
### Description of the bug

For the attached simple symnmetrical widget, "support on built plate only" generates a first layer support in the clear hole. There's no reason to generate first layer supports in the hole.

[image]
Angled view of model

[image]
Top view of issue (support on build plate only)

[image]
Top view of supports "Everywhere", which doesn't show this issue

### Project file & How to reproduce

[latch hub.zip](https://github.com/prusa3d/PrusaSlicer/files/9855931/latch.hub.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 11 22H2

### Printer model

MK3

neophyl, 2022-10-25:
PS is doing exactly what its supposed to do with the settings configured in your project.  

You have Print Settings>Support material>First Layer Expansion set to its default of 3mm.  That will extend the first layer of the support outwards by 3mm in all directions.  Also you are using Grid type supports which will further expand the support itself.

Change the type to Snug and set the first layer expansion to 0 and you get
[image]

brianbienvenu, 2022-10-25:
Thanks neophyl,

Those are the default values, I haven't played with them.

If it's an intentional feature to extend that inwards by 3mm, then why isn't the expanded section connected to the rest of the "raft" as the outer expansion is? Why is this the First Layer Expansion not applied equally to both types of support ("Everywhere" and "Support on build plate only")?

neophyl, 2022-10-25:
Good questions, I have wondered about that with this example model too.  I don't have a good answer though :)  The extending inwards is normal, its just a case of why its treating 'on build plate' and 'everywhere' differently with this model as on the surface there doesnt appear to be a different support structure.

brianbienvenu, 2022-10-25:
I suppose I found it weird that it was generating a support raft that didn't connect with anything at all. Seems like a waste of filament. I realise that it's about 0.01g of filament 😃

vovodroid, 2022-11-09:
And what's wrong in this case with snug support?

[image]

Grid does produce support column:

[image]

[Nozzle-TubeNozzle.zip](https://github.com/prusa3d/PrusaSlicer/files/9971610/Nozzle-TubeNozzle.zip)

neophyl, 2022-11-09:
I don't think there's anything 'wrong' with it really.  The area doesn't actually need any support in the first place.  There's a very small area detected as dark blue so potentially could be supported but in reality supporting would be a waste of time and material.  
Grid support always used to massively expand outwards the supported area lots, that was one of the most complained about aspects of Grid and a driving for force for the Snug option.

Snug basically is detecting that as very small and so not worth building a support for.  If you set the Closing Radius for snug to something like 4-5mm then it too will expand outwards enough to make it worth actually building the support upwards.
What the current Snug support  algorithm does have a problem with is that its not doing a check as to when it doesn't build support it is still doing the first layer expansion part.  It should really only do that part after its decided if its going to actually do the support column.

vovodroid, 2022-11-09:
Nevertheless printing small pad without any goal is bug anyway.

neophyl, 2022-11-09:
Isn’t that what I said ?  It’s doing the first layer expansion even once it’s decided not to generate the support.

DragonFinlay, 2022-11-15:
#9152 has supports where they are not needed, and it's only solved by decreasing the closing radius to 0.
Even then, the first layer includes the "base" for supports that would be there, which can then be removed by setting the first layer expansion to 0.

brianbienvenu, 2026-09-29:
This may not be a critical bug, but this exact issue is still present in 3.0.0 alpha 12:

[image]

There is support material being printed on the first layer in the middle of the shaft, that doesn't connect with anything.