prusa3d/PrusaSlicer #9089: Ooze Prevention w/ Wipe Tower

DegreelessEngineering, 2022-10-25:
**Is your feature request related to a problem? Please describe.**
I'm using PrusaSlicer with a Raise3D Pro3. When printing with both extruders, it won't allow me to use a Wipe Tower AND Ooze Prevention (Error: Ooze prevention is currently not supported with the wipe tower enabled). 

**Describe the solution you'd like**
Allow both to be used simultaneously. Consider moving/adding ooze control temperature setting/variation to Filament Settings page under Filament or Filament Overrides.

**Describe how it would work**
IdeaMaker already utilizes both Wipe Tower and Ooze Prevention simultaneously. **I've attached a screenshot of their settings page** By default they have "Cool Down Inactive Extruder" enabled with a 25degF drop when unused (PLA). They also have an option "Heat up Inactive Extruder in Advance" function which is pretty cool and you can even dial in the "Heat up Ahead of Time". This function is disabled by default, but a nice function no less. 

**Describe alternatives you've considered**
I have already copied the Retraction Speed (20mm/s), Deretraction Speed (20mm/s), and Retraction when tool is disabled (advanced settings for multi-extruder setups) to 11mm and Extra Length: 0 from IdeaMakers PLA Profile over to PrusaSlicer and it's not quite enough.

**Additional context**
Add any other context or paint drawing about the feature request here.
[image]

h8nor, 2022-11-01:
Two tool-heads printer with similar operation of extruders: https://github.com/prusa3d/PrusaSlicer/issues/

lukasmatena, 2023-02-01:
Ooze prevention was improved in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2), including this. Feedback appreciated. Closing.

closed 2023-02-01 by lukasmatena