prusa3d/PrusaSlicer #9090: Initial homing has extrusion on during movement to priming strip for CR-1S Pro V2 profile

henryhbk, 2022-10-25:
### Description of the bug

The Homing sequence returns to the center point prior to moving over to the left to start the priming strip after automatic bed-level on the CR-10S. Once it moves to the center point it goes straight to the left and should be a "rapid" (e.g. G0 X0Y150) but it appears to be a tool move instead of rapid (so leaves an extrusion trail that joins the purge strip on the left side). This of course is in the middle of your model (assuming you are centered on the bed). This is quite annoying, to have to stand there to pull the strip off prior to the purge strip completing to not end up in your print.

### Project file & How to reproduce

Just do a print with the stock CR-10S Pro profile (the strip does not show in the preview)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

MacOS 12.6

### Printer model

Creality CR-10S Pro V2

pmjdebruijn, 2022-10-25:
@henryhbk:
Is your CR-10S Pro V2 on the latest firmware?
Can you share some photos or video of what is happening exactly?

henryhbk, 2022-10-25:
> @henryhbk:
> Is your CR-10S Pro V2 on the latest firmware?
> Can you share some photos or video of what is happening exactly?

I am running 1.7.0 of the firmware

[Video (sorry the auto level is painfully long)](https://vimeo.com/763953142/1fd602fc34)

pmjdebruijn, 2022-10-25:
Thanks for the video, I'll try to take a closer look tomorrow night.

However, at first glance, are you sure that's not just oozing? 
Maybe you're printing too hot? Or your filament is waterlogged?

henryhbk, 2022-10-26:
I mean I have printed thousands of prints, and never had that with any of my 3 prusa printers, gcreate or my taz-6. Going to try a test print with simplify3d to make sure it’s not a firmware bug. And no this is not oozing, the extruder is turning during the move to the left. I assume the move to the left in the g-code will not be a g0, I will have to look. It is 100% reproducible

henryhbk, 2022-10-26:
Just for testing I just did a similar job in Simplify3D and it does not exhibit this behavior (so the behavior is in the generated g-code). Of note, simplify 3D doesn't do a priming strip at all (just uses the skirt as priming)

henryhbk, 2022-10-26:
Oh, and as for waterlogged that filament was right out of the fresh opened bag, and the rest of the print goes just fine. If the hob wasn't turning all those would be reasonable thoughts, but given the hob is in fact turning, that is a commanded extrusion.

pmjdebruijn, 2022-10-26:
Thanks for the video, however you turn away to the control panel, at a crucial point. Could you make a new video more zoomed out, so most of the printer remains in view the whole time, possibly at a slightly lower angle, so we can see the nozzle?

Can you also post the actual start_gcode from the generated .gcode file you're having this issue with? Because the behavior as far as I can see makes no sense. So either the generated .gcode is defective, or we're experience some bizarrre firmware bug.

henryhbk, 2022-10-26:
[E3D giant collet.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/9873715/E3D.giant.collet.gcode.txt)
changed the extension to be .txt for GitHub compatibility.

pmjdebruijn, 2022-10-27:
Could you try replacing the start_gcode with the following, and see if the issue persists:

`
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S150 ; set temporary nozzle temp to prevent oozing during homing and auto bed leveling
M140 S{first_layer_bed_temperature[0]} ; set final bed temp
M190 S{first_layer_bed_temperature[0]} ; wait for bed temp to stabilize
G28 ; home all axis
G29 ; auto bed levelling
G1 Z50 F240
G1 X2.0 Y10 F3000
M104 S{first_layer_temperature[0]} ; set final nozzle temp
M109 S{first_layer_temperature[0]} ; wait for nozzle temp to stabilize
G1 Z0.28 F240
G92 E0
G1 X2.0 Y140 E10 F1500 ; prime the nozzle
G1 X2.3 Y140 F5000
G92 E0
G1 X2.3 Y10 E10 F1200 ; prime the nozzle
G92 E0
`

pmjdebruijn, 2022-10-31:
The changes above should now be available in version 0.2.3 of the Creality bundle.

rtyr, 2022-11-01:
I am closing this issue.

@henryhbk We can reopen if there are still some issue.

closed 2022-11-01 by rtyr

sparklesonthegrid, 2023-04-28:
I'm having this exact same issue on the CR-10S Pro V2 profile. Prusa Version 2.5.2+win64. Printer firmware version 1.70.0. Even if I wait around and pull the bit off the nozzle before it lowers all the way to the plate, it still leaves a line of filament across the middle of the build plate.

My start gcode is the same as what was pasted in the Oct 27, 2022 comment. This has been happening since I got my printer a couple months ago, so I can't compare it to a time when this wasn't happening. 

The video henryhbk posted shows the exact issue I'm having. 

It feels like the first "prime the nozzle" command is the culprit, but today is the first day I've tried to interpret gcode so I could be wrong. Turns out just because it's "code" doesn't mean my computer science degree helps me a bit when figuring out what's going on.

sparklesonthegrid, 2023-04-28:
I was right and wrong. the prusa preview of what the nozzle location was didn't match what was actually happening. Watching my next print, "G1 X2.0 Y10 F3000" line does execute (after auto bed level), but then the nozzle goes to the middle of the build plate when setting final nozzle temp and waiting to stabilize. 

At this point the slicer think the nozzle is at 2,10, but really it's at ~150,150. So the first prime the nozzle command wipes it through the middle of the build plate, instead of along the side like Prusa slicer thinks it should. 

For my next print I'll try moving the "G1 X2.0 Y10 F3000" line to after the final nozzle temp heat and see if that fixes it.