prusa3d/PrusaSlicer #9092: gcode does not include temperature.

Z28Tempest, 2022-10-26:
### Description of the bug

I triple checked that nozzle and bed temperatures are set in the PuraSlicer settings. 
I sliced a print, exported to gcode, copied to SD card, and ran it. 
My Creality CR-10 V3 never gets the temperature from the gcode.  

### Project file & How to reproduce

[ksr test_v4.zip](https://github.com/prusa3d/PrusaSlicer/files/9865440/ksr.test_v4.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

W11

### Printer model

Creality CR-10 V3

neophyl, 2022-10-26:
Your saved project settings seem a bit unusual.  For example you have gcode flavour set to Mach3/LinuxCNC which does seem to then omit the temperature commands.  Not surprisingly.

Also its usual to have a block in the start gcode to set filament and bed temperatures.  If you don't then PS will add them automatically (for most gcode flavours) but then you dont get to control where and when they are added.

Something like
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

is common.  There is very little in your start gcode configuration really.  Normally you would also expect to see commands to set if absolute or relative positioning is being used etc too.

Basically it looks like your profile is based on a default custom and not the included Cr10-V3 profile you can install from the wizard as it shows up as a printer type of "My Settings (modified)" when the project is opened.

Nailig, 2022-10-27:
I agree, the configuration is not correct.
Most obvious is the wrong gcode-variant, it should be Marlin (legacy), not Mach3/LinuxCNC (CNC machines do not use temps)

If i load your file and use the default Creality CR-10 V3 config from prusaslicer 2.5.0, the temperatures are set.
Use the configuration-wizzard to include your printer and use that profile.

rtyr, 2022-10-27:
Answered. Closing.

closed 2022-10-27 by rtyr