prusa3d/PrusaSlicer #9094: Speed Modifier doesnt apply properly on several sections

SeanLangre, 2022-10-26:
### Description of the bug

Speed modifier doesn't apply on internal infill, solid infill, top solid infill etc.

Im trying to lower the speed on the edges of my model but I cant seem to override some of it. 

Speed is 140mm/s, speed I want is 25mm/s.

[image]
[image]

### Project file & How to reproduce

Open project, slice, and check View -> Speed.

[speed-modifier-bug-test.zip](https://github.com/prusa3d/PrusaSlicer/files/9872500/speed-modifier-bug-test.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10

### Printer model

Prusa Mini+

kubispe1, 2022-11-07:
Hi, thank you both for comments. I see this issue is here for a such a longer time #4202. 
Reason is that editing infill makes others defects. So this option is here, but not allowed internally.

DragonFinlay, 2022-11-10:
Nice find, I'll close this one as duplicate.

closed 2022-11-10 by DragonFinlay