prusa3d/PrusaSlicer #9095: V 2.5.0 sélection extrudeur multiple

18pool49, 2022-10-27:
### Description of the bug

les fonctions séquence d'extrudeur  ne fonctionne pas du tout ( le pignon en bas ) on arrive difficilement a faire des bandes de couleurs mais quand on tranche il y a plus de bandes de couleur
[image]
[image]

### Project file & How to reproduce

[cycloidic_vase.zip](https://github.com/prusa3d/PrusaSlicer/files/9880980/cycloidic_vase.zip)
une fois la sélection des bandes faite  
[image]
[image]
et que l'on tranche ; toutes les bandes disparaissent

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 

### Operating system

W10 win 64

### Printer model

A10M geeetech

18pool49, 2022-10-27:
la version précédente fonctionnaire très bien ; V2.4

18pool49, 2022-10-28:
comme je l'ai écrit la version 2.4 fonctionnait bien , donc je suis retourné sur cette ancienne version , et a ma grande surprise j'ai le MÊME problème la même panne et je viens de me rappeler que la seule chose qui est changé sur mon ordi C'EST LA MISE A JOUR DE W10 22H2 
[image]
car avant cette mise a jour tout fonctionnait correctement avec la V 2.4

18pool49, 2022-10-30:
je viens de changer la valeur de mes rayures de 3mm>>  je veux 15mm et ca me donne rien sur le dessin ,je ne suis même pas dans la couleur prévu a droite ; je relance ma demande de rayures et surprise voila mes rayures sur le dessin et a droite ; mais quand je tranche mais rayures disparaissent et au lieu d'avoir une pièce >rouge demandé >>> elle est jaune ??? et pas de rayure 
[image]
je redemande mes rayures et les voilà qui sont bien présentes 
[image]
mais impossible de trancher les rayures disparaissent

DragonFinlay, 2022-11-02:
Hi, I'm sorry but none of us really speak French. I tried translating your posts but many things don't make sense because there are technical terms.
Could you please explain the issue in English?

18pool49, 2022-11-02:
the extruder sequence functions do not work at all (the pinion at the bottom) it is difficult to make color bands but when you slice there are more color bands
and that we slice; all bands disappear
when you want to make strips, no function works either with the gear down or the cross (+) on the slider up
when I can hardly make scratches, those disappear when I slice
and when I asked for my piece in red on the right part of the screen, once sliced it becomes YELLOW but without stripes
extruder changer not working

18pool49, 2022-11-02:
2 or 3 weeks ago version 2.4 worked fine; and the 2.5 too
but following the update of W10 22h2 nothing works anymore in extruder change

18pool49, 2022-11-02:
I can't change colors anymore

YuSanka, 2022-11-03:
I suppose, you forget to press "Reslice" button after change of the extruders sequence.

18pool49, 2022-11-03:
Nope !! no way !! I ask for the cutting with each modification; but the requested colors disappear and are not realized in the Gcode

18pool49, 2022-11-03:
the problem started to occur since version W10 22H2

18pool49, 2022-11-03:
but the requested colors disappear and are not realized in the Gcode >> when I manage to make color changes, >> because it does not work every time

18pool49, 2022-11-03:
the whole "change extruder" function is faulty

YuSanka, 2022-11-04:
Ok, I tried it on my Win10 laptop and can confirm this issue.

18pool49, 2022-11-04:
super ;; me voilà rassuré vous avez tout sous la main pour remédier au problème 
Great ;; I'm reassured you have everything at hand to remedy the problem

18pool49, 2022-11-06:
merci beaucoup a vous [YuSanka](https://github.com/YuSanka) qui avez résolu ce problème 
ca fonctionne parfaitement

YuSanka, 2022-11-07:
Let's keep it open until we fix it ;)

18pool49, 2022-11-07:
ha ok because I haven't noticed any new anomaly, for me, everything seems to be back to normal, because I did my tests

kubispe1, 2022-11-07:
And what kind of test ? Or what have you changed? :D

18pool49, 2022-11-07:
how do you remove "place on tray"
[image]
to make two tubes on the bed at the same time
a Ht.150 tube: cut at the Ht.50 in blue, and the rest 100mm in red
a tube of ht 150: cut at the Ht 100 in red, and the rest 50mm in blue

closed 2026-09-29 by prusaslicer-bot[bot]