prusa3d/PrusaSlicer #9097: Make elephant foot compensation available to modifier settings

jphalip, 2022-10-27:
There currently doesn't seem to be a way to change the elephant foot compensation on a modifier.

It'd be great if this could be added to the list of available settings when you right-click on a modifier and then select "Add settings". It could perhaps go into the "Advanced" category.

neophyl, 2022-10-27:
You already can.  Select the object, click on The editing icon in the tree view on the right and a list appears.  At the very bottom of the list is Add Settings.  Select that and click advanced.  Elephants foot is in there.  

The modifier options in the Infill/Layers and Perimeters/Support Material are only the basic ones.    Selecting Advanced gives you a lot more to select from.  Its been that way for ages.  The same works for modifiers too, once you have a modifier there are many more options if you use Add Settings.

jphalip, 2022-10-27:
@neophyl That's odd. I was already aware of the "Add settings" menu that you mention. However, under "Advanced" I only see the following options: Bridge flow ratio, Length of the infill anchor, Maximum length of the infill anchor, and Infill/perimeters overlap.

Do you know why I can't seem to see all the other settings there? Do I maybe have to activate/enable something?

neophyl, 2022-10-27:
Not really.  The Add settings option isnt there at all if you arent in expert mode so I dont know why you dont have it
Heres what is under Advanced for me
[image]

jphalip, 2022-10-27:
Ok, so I figured it out! I was clicking on a "part", i.e. a sub-part of an object. Apparently the options there are limited. But if I try on a main object, then I can see all the options that you mentioned.

closed 2022-10-27 by jphalip