prusa3d/PrusaSlicer #9099: GCode View distorted when opening File from https://fullcontrol.xyz/

mburger82, 2022-10-28:
### Description of the bug

When I create a "Pin Support Challenge" File for CR10 on https://fullcontrol.xyz/ and open it in PrusaSlicer, the Printing bed is drawn distorted. Also PrusaSlicer Controls ar not drawn properly. 
Example GCode generated here: https://fullcontrol.xyz/#/models/67cf20

[image]

### Project file & How to reproduce

Example GCode File:
[Pin-Support Challenge_D2022-10-28_T09_41_49.zip](https://github.com/prusa3d/PrusaSlicer/files/9887316/Pin-Support.Challenge_D2022-10-28_T09_41_49.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Win10

### Printer model

cr10

mburger82, 2022-11-02:
Thanks for the reply.
I understand, that the experimental G-Code results in strange rendering. 
But what about the scrambled right side of the screen? This seams not intended and presumably "buggy"

DragonFinlay, 2022-11-04:
> But what about the scrambled right side of the screen? This seams not intended and presumably "buggy"

Sorry, didn't even see that. Thanks for reporting!

kubispe1, 2023-02-01:
Can we close this issue ? @mburger82 Can you check it, please?

mburger82, 2023-02-02:
@kubispe1 
Yes, just tested it with 2.6.0alpha. Seams to work. thx.

closed 2023-02-02 by mburger82

kubispe1, 2023-02-07:
Fine, thanks