prusa3d/PrusaSlicer #9100: Search feature in the settings is broken - every time moves the cursor at the end of the text

Eldenroot, 2022-10-28:
### Description of the bug

Use search function in the settings menu is bugged - you cannot type into the already typed text.

### Project file & How to reproduce

Open search function, type some text (use search in English, I use Czech language) ... now try to type some text into the already typed text -> you are every time moved to the ned of the text automatically.

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 11 64bit

### Printer model

Prusa Mini+

DragonFinlay, 2022-11-02:
I can't reproduce it, it works fine for me:

[image]

YuSanka, 2022-11-03:
@Eldenroot , thanks for your report. I can reproduce your issue. @DragonFinlay , issue appears in SearchDialog from Preset Tab.

YuSanka, 2022-11-03:
This issue is reproducible from first implementation (even in 2.3.0)

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena