prusa3d/PrusaSlicer #9101: start_gcode before the hot bed turns on

yeyun1999, 2022-10-28:
**Is your feature request related to a problem? Please describe.**
I am using klipper and have set a pin to control the relay of the switching power supply. I need to turn on the switching power supply before starting to print, because the custom start_gcode is after the  turned on hot bed.I wish I could put the gcode that turns on the relay before the hot bed turns on。

neophyl, 2022-10-28:
You can.  PS only add the bed heating commands automatically if you don't add them to your start gcode yourself.  

So add a block that turns on your relay and then add the appropriate bed heating lines.  That way you control when and where they appear.

yeyun1999, 2022-10-28:
> You can. PS only add the bed heating commands automatically if you don't add them to your start gcode yourself.
> 
> So add a block that turns on your relay and then add the appropriate bed heating lines. That way you control when and where they appear.

My problem has been solved, thanks.

closed 2022-10-28 by yeyun1999