prusa3d/PrusaSlicer #9102: modifier with fuzzy skin splits part

Wanderwardr, 2022-10-29:
### Description of the bug

when a modifier is applied to a part with a fuzzy skin property, the fuzzy skin gets extended into the internals of the part and splits it (image attached).  I expected the fuzzy skin to stop at the edge of the surface of the part at the edge of the modifier and then have the surface of the part continue to the smooth region, but it looks like it extends into the part along the modifier.  The images of the attached part have a modifier that extends into the part.  This same part has a second modifier applied that mostly covers the interior surfaces, but the same issue did not occur there.
[image]

[image]

### Project file & How to reproduce

Here is the project I am working on, along with the amf file used to generate it. The amf file is also used as a modifier to fit the contoured shape of the part so that the exterior surface remains smooth.
[FWD Tray DEEP.zip](https://github.com/prusa3d/PrusaSlicer/files/9892717/FWD.Tray.DEEP.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0, however the issue also occured on 2.4.2

### Operating system

10

### Printer model

Prusa Mk3S

neophyl, 2022-10-29:
Many of the modifier properties split to different volumes not just fuzzy skin. That’s the way they work (currently). You will find many similar issues raised if you search (both open and closed).

DragonFinlay, 2022-11-03:
What neophyl said. We have many issues about that and we have it on the radar.

Closing as duplicate of #8121.

closed 2022-11-03 by DragonFinlay