prusa3d/PrusaSlicer #9109: Slicer can't finish multi-material on the printer with 2 extruders

h8nor, 2022-11-01:
### Description of the bug

The printer with [2 tool-heads](5916#issuecomment-769463509). Set value as specified in [recommendation](7514#issuecomment-996919224). I tried to adding second extruder on the Prusa i3 MK3S printer. The result is negative.

The Cube3 with two tool-heads printer has proprietary software. Some commands are not recognized by the PrusaSlicer G-code Viewer[*](6772), but it are not reported in an error. The official version of the program Cube Print 4.03 had the ability to create g-code for grouped STL files.

[PrusaSlicer.webm](https://user-images.githubusercontent.com/6163937/199180080-97575b9e-f407-4eff-a298-e99ad4d1e1da.webm)

### Project file & How to reproduce

I am attaching the original STL file below so that I can reproduce the error.
1. Added a second extruder, given color
2. Filled in the `toolchange_gcode` field to check the generated g-code
  ```ini
{if next_extruder == 0}
; EXTRUDER:0 LAYER:{layer_num} Z{digits(layer_z, 0, 3)}
{endif}
{if next_extruder == 1}
; EXTRUDER:1 LAYER:{layer_num} Z{digits(layer_z, 0, 3)}
{endif}
  ```
3. Import [(vase_faceted.STL).zip](https://github.com/prusa3d/PrusaSlicer/files/9907728/vase_faceted.zip) file
4. Split the STL file into parts (as shown in the video)
5. Assign extruders to each part (as shown in the video)
6. Opened the created g-code and try to find
  ```ini
; EXTRUDER:1
  ```
Cannot find a match. Also visually the program resets the second extruder.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-win64

### Operating system

Windows 10 21H1 Build 19043

### Printer model

Cube 3

rtyr, 2022-11-01:
Please provide your project file.

h8nor, 2022-11-01:
_Added a project file to the Checklist section._

h8nor, 2022-11-01:
_Added the project file to Checklist section._

h8nor, 2022-11-01:
Thank you, done. In regex didn't set parameter Much single line. `[Slic3r_PE.config: 115, 35]`

Can the PrusaSlicer G-Code Viewer somehow display a line number because of which I cann't open a some third-party g-code?

h8nor, 2022-11-01:
Thank you, resolved. In regex didn't set parameter Match single line. `[Slic3r_PE.config: 115, 35]`

Can the PrusaSlicer G-Code Viewer somehow display a line number because of which I cann't open a some third-party g-code?

h8nor, 2022-11-01:
The error in slicer is associated with the compelled commenting of the command `T0` and `T1`.

rtyr, 2022-11-01:
G-code previewer previews actual g-code, so it is expected behavior caused by your g-code substitution.

h8nor, 2022-11-01:
I understand that your printers use open standards.
In the Cube3 the temperature on extruders set by commands `M104` and `M204`, and physically both nozzles are in assembly module. There are more expensive printers with analogue assembly like the [Raise3D Pro3](8894).

Could you use commands `T0` and `T1` in field `toolchange_gcode`?

rtyr, 2022-11-01:
> Also visually the program resets the second extruder.

This is caused by your regex. G-code preview shows the final g-code. Standard command for toolchange is Tx (T0, T1,...) for Marlin (your selected g-code flavor). No T-command, no tool change.

> Cannot find a match.

That is caused by a typo in your toolchange script.

[image]

h8nor, 2022-11-01:
Yes, the **Project file** really contains a typo in the indicated place. Since the settings have been copy paste.

The Cube3 does not use command `T0`, instead it needs to perform the code block `{if next_extruder == 0}...{endif}`. Similarly with the command `T1`. Code blocks differ.

Other indicated commands are excluded because they can violate a print. In fact with settings PrusaSlicer from the **Project file** may will incorrectly display a model, but this will not affect printing.

rtyr, 2022-11-01:
Yes, but I still don't understand why the issue was reopened by you, I don't see any problem.