prusa3d/PrusaSlicer #9111: Please add an option to keep Paint-on supports, Seam painting and Variable Layer Height after Cut

laptop-laptop, 2022-11-01:
**Is your feature request related to a problem? Please describe.**

I applied Paint-on supports, Seam painting and Variable Layer Height. Then, I executed a Cut.
After that, all those applications disappeared, and I have to do the entire things all over again.

**Describe the solution you'd like**
Not sure why such modifications disappeared after Cut was executed. 
Perhaps add an option to keep these modifications even after Cut is executed.

DragonFinlay, 2022-11-04:
Your idea sounds simple but is likely very difficult to execute. You have to keep in mind that by slicing the object it has to be remeshed, meaning that the geometry changes. This means that the coordinates for the paint-on features are lost.
Perhaps we can do something about this in the future, but it's probably not a priority right now.

rcmaniac25, 2023-03-03:
This almost bit me when I wondered why a 12h print was going to be so much faster... and then I noticed the paint-on supports had disappeared. So would very much like this at some point.

rcmaniac25, 2026-09-13:
Still an issue.