prusa3d/PrusaSlicer #9113: Arrange doesnt work correctly

aclucca, 2022-11-01:
### Description of the bug

Function "Arrange" dont work on this Skull.
(Spacing is Set to 0)

If i rotate Z 30 degrees it will fit . . . .  

### Project file & How to reproduce

[image]
[image]
[fancyskull1-repaired_rotat.zip](https://github.com/prusa3d/PrusaSlicer/files/9914003/fancyskull1-repaired_rotat.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10 64bit

### Printer model

Prusa MK3

foreachthing, 2022-11-02:
Right-click on arrange and enable rotation.
Then it should work.

aclucca, 2022-11-02:
I´m sorry - but enabling rotation doesnt fix this issue.

ulab, 2022-11-26:
Another example:

Zack Freedman's "Nipper Holder x6" fits on the Mini's bed after rotating 45°, but arranging will give this error even with rotating enabled and even when it obviously already fits.

[image]

You can find the object to download on Thangs:

https://thangs.com/designer/ZackFreedman/3d-model/Gridfinity%20Nippers%20%2B%20Strippers%20%2B%20Pliers%20Racks-61120

stepikovo, 2022-12-15:
I'm closing this as a duplicate of #5339

closed 2022-12-15 by stepikovo