prusa3d/PrusaSlicer #9114: Simple "find and replace" option as post-processing

Tupson444, 2022-11-01:
Sometimes I want to alter the Start G-code (or something else) for one print only. Of course, I can change it in the Printer Settings or Filament Settings, but I would like to save it within the Print Settings.

**Describe the solution you'd like**
A good solution would be to use post-processing to find and replace text, right under Print Settings > Output options, without using external scripts. As I can see, that is possible in Simplify3D (for example {REPLACE "old text" "new text"}), but it doesn't work in PrusaSlicer (error: The configured post-processing script does not exist). Am I just missing something?

**Describe how it would work**
I can place a comment in Start G-code, for example ";find-replace:start". Then use find and replace as post-processing to replace ";find-replace:start" with whatever G-code needed and save Print Settings (without the need to save separate Printer Settings preset).

**Describe alternatives you've considered**
For now I can export the G-code and use any text editor to manually find and replace text.

Or I can use filament-specific start G-code and save Filament Settings (in which case I need to save separate Filament Settings even though all other filament-related settings are the same).

bubnikv, 2022-11-02:
PrusaSlicer supports find and replace using regular expressions since
PrusaSlicer 2.4.1 I believe. The feature is certainly supported by the
currently released PrusaSlicer 2.5.0.

Tupson444, 2022-11-02:
Oh sorry, I didn't find it before, I tried typing under "post-processing". Now I see it's in "G-code substitutions". My bad, I'm closing the issue.

closed 2022-11-02 by Tupson444