prusa3d/PrusaSlicer #9119: painted areas are not the right color after slicing.

huddlesj, 2022-11-03:
### Description of the bug

I am trying to make a multicolor version of the rugged battery box ( https://www.thingiverse.com/thing:4966369 ) from Thingiverse. 

I have it painted in two colors 

[image]

After it is sliced both items on the bed have a spot on the right front that is the wrong color 

[image]

[AA_-_Bottom.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9930497/AA_-_Bottom.3mf.zip)

### Project file & How to reproduce

1. start new multicolor project
2. import AA battery box top or bottom
3. Use paint tool smart fill to change the raised areas a different color
4. Slice
5. Check all raised edges to make sure they are the right color

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+MacOS-x64-202209060719

### Operating system

macOS 13.0

### Printer model

Original Prusa i3 MK3S MMU2S

DragonFinlay, 2022-11-08:
I can reproduce it and can't see an easy way to fix it. We'll take a look, thanks for reporting!

kubispe1, 2022-11-09:
Hi, I have repaired both models and paint critical areas and I think it is ok.
[image]
3mf with partial paint:
[AA_-_Bottom_REPAIR.zip](https://github.com/prusa3d/PrusaSlicer/files/9968872/AA_-_Bottom_REPAIR.zip)

@huddlesj Can you check please ?

huddlesj, 2022-11-18:
@kubispe1 The issue is fixed in that file. Thanks, Jay

DragonFinlay, 2022-11-18:
Issue with damaged model, closing.

closed 2022-11-18 by DragonFinlay