prusa3d/PrusaSlicer #9128: Slicer is crashing when generaing G-code

patryk-mol, 2022-11-06:
### Description of the bug

Slicer is crashing when generaing G-code for attached stl.

### Project file & How to reproduce

Import the attached stl file and generate G-code. [monitor-stand-half.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/9945608/monitor-stand-half.stl.zip)

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Arch Linux, kernel 6.0.7-arch1-1

### Printer model

MK3S+

PhilQ-github, 2022-11-06:
Crashes for me too, doesn't seem to matter what the model is (using default settings). Also Arch Linux kernel 6.0.7-arch1-1. Output on terminal shows an assert issue....

/usr/include/c++/12.2.0/bits/unique_ptr.h:445: typename std::add_lvalue_reference<_Tp>::type std::unique_ptr<_Tp, _Dp>::operator*() const [with _Tp = Slic3r::SpiralVase; _Dp = std::default_delete<Slic3r::SpiralVase>; typename std::add_lvalue_reference<_Tp>::type = Slic3r::SpiralVase&]: Assertion 'get() != pointer()' failed.
Aborted (core dumped)

Fixed by reverting to prusa-slicer 2.4.2-9 package which was still in the package manageer cache (/var/cache/pacman/pkg) directory.

kuon, 2022-11-06:
Same error here, you can revert to `2.4.2` with (even if not in your package cache):

```
sudo pacman -U https://archive.archlinux.org/packages/path/prusa-slicer-2.4.2-9-x86_64.pkg.tar.zst 
```

CountMurphy, 2022-11-06:
Also running Arch (kernel 6.0.7). Prusa slicer crashes with any model.

feythr, 2022-11-06:
Looks like the same issue as #8981 (the packager disabled that test in the PKGBUILD https://github.com/archlinux/svntogit-community/commit/0b21c0b43d7f6e622825985eef52d7e7aa5ec6bf, but that doesn't fix the underlying issue). Seems like there's a difference in compiler behavior that affects when the lambda captures in `process_layers` are evaluated.

feythr, 2022-11-07:
@svenstaro Can we apply 4b686af12db49500205ca15038fb7d651ccc4295 to the Arch package (and re-enable `fff_print_tests`)?

svenstaro, 2022-11-07:
Backported this into Arch. Tested it and it's working.

lukasmatena, 2022-11-08:
Ok, closing.

closed 2022-11-08 by lukasmatena

PhilQ-github, 2022-11-08:
This bug need to be reopened. Although the immediate slicing now works (good step), if you try to use the slider to examine the layers sliced, it crashes again. Probably the same problem elsewhere in the code.

PhilQ-github, 2022-11-08:
Here is the output on the console:
/usr/include/c++/12.2.0/bits/unique_ptr.h:445: typename std::add_lvalue_reference<_Tp>::type std::unique_ptr<_Tp, _Dp>::operator*() const [with _Tp = Slic3r::SpiralVase; _Dp = std::default_delete<Slic3r::SpiralVase>; typename std::add_lvalue_reference<_Tp>::type = Slic3r::SpiralVase&]: Assertion 'get() != pointer()' failed.
Aborted (core dumped)

lukasmatena, 2022-11-08:
That is quite suspicious. Unless I am mistaken, dereferencing `std::unique_ptr<SpiralVase>` does not happen anywhere in the code after the mentioned patch is applied. Are you absolutely sure that you are indeed running the updated package? Can anyone else reproduce it? If so, is anyone able to provide a complete backtrace of the abort call? Thanks.

svenstaro, 2022-11-08:
So I just sliced something with my new package and investigated the layers and added colors and what not and couldn't get it to crash.

PhilQ-github, 2022-11-08:
I just took thr 3dBenchy .stl from Prusa's printables site (https://www.printables.com/model/3161-3d-benchy/files) and was able to slice it normally. Before the patch the slice operation failed with the crash, so it must be the new package. As soon as I try to move the slider down to show the layers, it crashes.

If somebody can tell me how to produce the complete backtrace, I will post it.

PhilQ-github, 2022-11-09:
Looks like the Arch package has been reworked since yesterday, now as prusa-slicer 2.5.0-3. The issue does not show with this version for me.

svenstaro, 2022-11-09:
2.5.0-2 should already have been fine.

PhilQ-github, 2022-11-09:
I wonder if it was just the mirror site I am using that was lagging behind?

The PKGBUILD is not working for me, I tried pulling it from the Arch Build system with the intention of tryinmg to build it myself and it is complaining about a missing patch file:

==> ERROR: prusa-slicer-fix-lcereal-p1.patch was not found in the build directory and is not a URL.

svenstaro, 2022-11-09:
Are you sure your abs is updated? At any rate, let's not do arch specific stuff here in their tracker.

fordfrog, 2023-03-19:
i tried both 2.5.0 & 2.5.1 and having that issue with the following project: [test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11011316/test.3mf.zip). i did not test other projects though.

the error:
```
/usr/lib/gcc/x86_64-pc-linux-gnu/12/include/g++-v12/bits/unique_ptr.h:445: typename std::add_lvalue_reference<_Tp>::type std::unique_ptr<_Tp, _Dp>::operator*() const [with _Tp = Slic3r::SpiralVase; _Dp = std::default_delete<Slic3r::SpiralVase>; typename std::add_lvalue_reference<_Tp>::type = Slic3r::SpiralVase&]: Assertion 'get() != pointer()' failed.
Neúspěšně ukončen (SIGABRT)
```

i'm on gentoo linux with kernel 6.2.7.

fordfrog, 2023-03-25:
today i compiled 2.6.0_alpha5 and it does not crash when slicing the same project so the issue is there, it's just not backported.