prusa3d/PrusaSlicer #9129: quick slicer without saving project.

antiairdevil, 2022-11-06:
**Is your feature request related to a problem? Please describe.**
I think there should be an option to not save the project when quiting slicer. Find it super annoying even if a have a clean slicer i need to save it to some project.

**Describe the solution you'd like**
maybe if you push i dont want to save the program can shut down without saving and not just stay alive and wait for saving.

**Describe how it would work**
make a option to quit without saving? or just make it shutdown without saving if you answer no to "do you want to save"

YuSanka, 2022-11-06:
Do you know about Checkbox "Ask for unsaved changes in project" in Preferences dialog? Just uncheck it ;)

antiairdevil, 2022-11-06:
Did not, but that will solve my problem, thanks!

closed 2022-11-06 by antiairdevil