prusa3d/PrusaSlicer #9130: Feature request: minimum perimeter / area size for fuzzy skin to avoid it on embossed text

edwardhorsford, 2022-11-06:
I've got a model with fuzzy skin on the inside and outside. The model also has embossed text on the top. I'd like the interior walls of the text to not have fuzzy skin.

[image]

I can add a modifier to disable fuzzy skin, but this produces other artefacts at the border of the modifier.

With modifier:
[image]

As text is often made up of small perimeters, a possible solution is to have a minimum interior perimeter size for the fuzzy skin. This would let small details such as screw holes and text be free of fuzzy skin, whilst the larger model gets it as normal. I believe PrusaSlicer is already doing analysis on perimeters to detect interior and exterior surfaces - this could be an extension of that, measuring the total distance or perhaps the area of the enclosed perimeter.

foreachthing, 2022-11-06:
This https://github.com/prusa3d/PrusaSlicer/issues/6988 could become a solution.

edwardhorsford, 2022-11-06:
> This #6988 could become a solution.

Indeed! Or if https://github.com/prusa3d/PrusaSlicer/issues/6881 weren't an issue too.

I suspect for some models though, a minimum area would be very useful - think a large model with many screw holes. It would be a pain to need to use a modifier for many tiny holes.

foreachthing, 2022-11-07:
Just use a CAD software or something like Blender to create a modifier mesh. Then import that and change type to modifier. Better yet, create a multibody part. That way you don't have to align anything.

edwardhorsford, 2022-11-07:
@foreachthing modifier meshes [screw with perimeters](https://github.com/prusa3d/PrusaSlicer/issues/6881).

DejfRCR, 2022-11-07:
Or just paint the fuzzy (or unfuzzy) like supports, that would make it less painful than using modifiers.

Snuff1eupagus, 2023-02-24:
Are you interested in an education on modifiers?

edwardhorsford, 2023-02-24:
@Snuff1eupagus I observe that modifiers cause issues at the boundaries of the modifier - see my second screenshot. Do you have suggestions for avoiding this?

Snuff1eupagus, 2023-02-24:
Yes I do, it's currently 2:28 am here I have prepared an example file for you, and an explanation of several things for you, and people reading this in the future. I will post it to this Feature Request in the morning so I can be clear headed in my Instruction and opinion.

edwardhorsford, 2023-02-24:
I'd love to see an example of avoiding this with modifiers 👍 

Note I've tried making modifiers in the shape of the text or possibly slightly larger. My comment there would be that those are a pain to make, and still have issues with fuzzy skin at the boundaries of the modifier.

Even if it works though... I think this feature request would be good. A model could have *lots* of inset text or small features. In general the fuzzy skin probably doesn't make sense for these - being able to easily threshold it would be helpful in these cases.

Snuff1eupagus, 2023-02-26:
@edwardhorsford 

I apologize, After posting this My mentor contacted me, and made a very good point, this is not the place for education, but rather the place where educated people make requests. As such I agree with his argument. I have to ask that you make a request on the Prusa forums. If you don't feel inclined to do so I understand, and regardless of what I have  been taught, I will drop the subject. If however you are actually interested in learning about modifiers please post in the English section of the prusa forums [PrusaSlicer ](https://forum.prusa3d.com/forum/prusaslicer/)section of the forums and I will show you what you need to know. 

Once again Apologies however my mentor made a good point, and that prevents me from providing education here.

closed 2026-09-29 by prusaslicer-bot[bot]